dawsonjon/PicoRX #1: QSD Question.

boxerbomb, 2023-11-09:
Why is it that in this design you only have a single RF source going into the FST3253, while in other designs including your own, you also supply an inverted input signal to pin 7?

dawsonjon, 2023-11-09:
Interesting question, there are a few versions of the qsd out there. You might have been refering to this design that uses a transformer as a phase splitter generating a normal and inverted copy of the signal. https://github.com/dawsonjon/OpenXcvr/wiki/Hardware-Design#quadrature-sampling-mixer

In this design I am using a differencing amplifier to achieve the same effect, switching between the inverting and non-inverting inputs of the amplifier instead. I think that the two approaches give equivalent results. The advantage is that it removes a (hand wound and potentially harder to obtain) transformer. I like the simplicity, especially in a stripped down project like this one. There are some designs that use both approaches,  some even use 'proper' differential amplifiers too. I don't think that these changes will make very much difference to the performance though.

I got the idea from this project 
https://circuitsalad.com/2020/01/06/compact-si5351-based-sdr/ which is another very minimal receiver. Although this is one of the configurations in Dan Tayloe's paper https://www.norcalqrp.org/files/Tayloe_mixer_x3a.pdf.

boxerbomb, 2023-11-09:
Interesting. Thanks for the information, I might try out a few variations.

closed 2023-11-09 by boxerbomb