dawsonjon/PicoRX #2: Quadrature signals seem to fail on higher frequencies

w9ran, 2024-02-05:
First, thanks very much for this excellent project Jon!    I've built up a board and while it is working I've run into a couple of anomalies,  
Note:  I used the 74CBT3253C  as the FST3253 is obsolete and not sold by US distributors any longer, but they are equivalent parts.  (I also did not include the preamp and my antenna is coupled via 100 nF to RF_PREAMP and to the bias network via 10K). 

After establishing that it was working on lower bands,  I had poor results trying to tune the 20 meter band - strange noises but no signals at all.  Even when using a signal generator it was completely dead, although it retuning to the lower bands proved the receiver still worked there.   

The first thing I do with a QSD is verify that the quadrature clocks were in quadrature- and they were.  However I noticed when I had scope probes temporarily attached to LO_1 and 2, the receiver started to work on 20 meters.   In fact because they are on the corner of the PCB I found I could make the receiver come to life just by touching pin 1 with a finger!  I found experimentally that 33pF was the minimum needed capacitance to ground from either LO_1 or LO_2.    There is kind of an "uncertainty principle" involved since the act of attaching a scope probe to make measurements was changing the behavior of the thing being measured!   

My first thought was that the high-speed clock signals might need to be terminated but even with a 390 ohm termination resistor to ground, the same behavior persisted.  The QSD would only work when capacitance is added - I made these two 20 sec. videos to show this with both hand capacitance (via tweezer) even with termination resistors in place, and with the 30pF capacitor:   

https://share.icloud.com/photos/0f8v431GzQMLIJ-MR3gZhBL_w
(signal source is 50uV (S9) at 14.1 MHz)

I can hear signals on any frequency up to 12 MHz signal without the capacitor but it is required to hear anything above 13 MHz.   The effect is repeatable and when working, the receiver is able to hear a 2-3uV signal in SSB bandwidth.  There seems to be no gray area - either it works or it doesn't.   Scoping the LO_I pin of the QSD shows no different with and without the capacitor in parallel even though the signal can be heard to appear only with it as in the videos. 

Unless there is some subtle difference between 3253 ICs, I'm out of theories -  but I ran into similar issues with my homebrew copy of the uSDX on the higher bands.  It has PCB traces about 1" long connecting the Si5351 and 3253, which I noticed on other radios were made as short as possible.    I had better results when reducing the output current of the Si5351 from 8 to 2mA, all of which suggests a problem with signal reflection on the PCB traces.  If so, the solution would require a complete redesign of the PCB and possibly adding termination resistors.   It would be very helpful to have confirmation using your PicoRX PCB layout.

Within the expected limitations of the DSP filters, the performance is very good when it's working - audio quality through an amplified speaker is excellent.   But this clock business has me baffled!  I'd certainly be interested in knowing if I am alone in what I am seeing!   Thanks and 73,
Bob W9RAN

dawsonjon, 2024-02-06:
Hi Bob,
Thanks very much for your interest in the project, and for taking the time to feed back on your progress. I'm sorry to hear about the problem you have been having. It certainly sounds like a signal integrity issue. I came across similar problems in my first prototype, but I put that down to the long lead lengths and breadboard-style construction. I haven't had any issues since I moved to the PCB and I'm getting reliable reception well beyond 20MHz. 

One thing you could try is reducing the slew-rate and drive strength of the LO outputs. It sounds like it would be worth adding some pads for series/parallel termination in the next PCB revision, and the trace lengths could probably be significantly reduced. I could also potentially swap some pins around to make the routing easier.

I have only tried the FST3253 so far, but I have bought a couple of alternative devices to try out. I was considering QS3253QG8 or possibly 74CBTLV3253PW (but at a lower supply voltage). The 74CBT3253C seems like a good candidate too, it wasn't available at my local supplier but it seems like it would be worth trying to see if I can replicate the issue.

Kind Regards
Jon
2E0GXA

w9ran, 2024-02-07:
Thanks for your reply Jon!    I look at DigiKey as kind of the universal reference point since their products are available internationally, and if you sort the dual 4:1 mux/demux parts by 16 pin SO (for easy hand soldering) the 74CBT3253C is what comes to the top.    But TI also made an older "non-C" version in which the /OE pins control both muxes (no individual /OE for each).    The 74CBT3253C that I used on the PicoRx should have identical function to the FST3253, but I went back in my notes from 3 years ago and have posted a summary on my site here for reference:  http://tinyurl.com/3c2rhwyh

On the topic of parts,  I received a Last Time Buy notice on the entire LTC6226/7 family of op-amps, which are evidently being discontinued by ADI.     

I'll build another board with termination resistors in LO_1 and LO_2 (ref:  https://resources.pcb.cadence.com/blog/termination-resistors-in-pcb-design)   What really puzzles me is how my receiver works or it doesn't - no gray area where it seems degraded but still works.

I grabbed a couple of scope traces to show the LO waveform with and without the added capacitor:
[image]

I also see considerable jitter in the quadrature clock duty cycle,  varying from approx 45 to 55%.  I haven't tried to quantify the effect of this, but "to the ear" the performance compares well with other QSDs.

Bob

dawsonjon, 2024-02-08:
Hi Bob, 

Thanks very much for sharing these details, I would be very interested to know how you get on with the termination. Its a shame about the LTC6226/7, they seemed like a very useful part. I used an LT6231 in the prototype and that worked well too.

I would expect to see 8ns of Jitter, this is introduced by the PIO's fractional divider. For non-integer clock divisions it aligns to the nearest clock edge which in this case is 125MHz. It could be reduced by overclocking the pico, but I prefer not to do this if I can help it. Its quite hard to quantify the effect of the jitter analytically, we know that the the clock edge can be up to a clock cycle away from where it should be, but how often this happens will depend on the tuned frequency. Also the jitter has a periodic pattern which introduces spurs rather than appearing a phase noise, again the magnitude and position of the spurs depend on the tuned frequency. There will inevitably be some degradation in performance, but in practice it does still seem to be a useable results.

closed 2024-08-09 by dawsonjon