dawsonjon/PicoRX #3: Image/Alias signals

w9ran, 2024-02-08:
I've been hearing signals that should not be where they are heard, such as CW in the middle of the 20 meter SSB band.   I also hear what I know from experience to be the "inverted speech" of an opposite sideband signal.   By using online spotting tools like DX Cluster and RBN I can document this for you.    Here are two screen photos and a short video of the PicoRX alongside my Xiegu G90.   In one, both receivers are  tuned to the actual signal the station is transmitting on, 14.238 MHz USB.   But in one, the Pico Rx is tuned to 14.282 MHz LSB and the same signal can be heard equally well on the short video:   

https://share.icloud.com/photos/042VUSVM5L2BdTouaNjXQnCVA

Another example:   The ARRL station W1AW sends code practice daily on specific frequencies including 14.0475 in the 20 meter band.   Here is a short video showing the PicoRX hearing the W1AW transmission when tuned to 14.297 MHz.   I change the frequency on the G90 to change the pitch which makes it clear the two are receiving the same signal:

https://share.icloud.com/photos/04ad-Pb5m6aTgxTEyo_fJKxJg

The antenna in use is an all-band vertical and both receiver are connected via a splitter so the same signal is being presented to both receivers.   
73, Bob W9RAN

dawsonjon, 2024-02-08:
Hi Bob,
Thanks again for the very informative feedback. I think there are probably a couple of things that might be going on here. 

Firstly, in your example where a signal can be hear equally well at 14.238 MHz and 14.282 MHz, the signals are separated by 44kHz so are both almost certainly within the 250kHz bandwidth of the ADC, this combined with the fact that the signal has an equal amplitude at both frequencies is suggestive of an image. This might be consistent with a missing signal in either the I or Q input channel resulting in a loss of image rejection. Are you seeing signals of similar magnitudes at the I and Q input to the ADC?

Secondly, aliasing is a weakness of this design which I briefly discuss here: https://github.com/dawsonjon/PicoRX/wiki/Hardware-Design#qsd-detector-tayloe-detector The ADC is relying on the QSD to act as an anti-aliasing filter. The QSD is effectively a second order filter so the roll-off is only about 12dB per octave. An aliased signal in the second nyquist region would likely still be heard, but I would expect there to be a noticeable attenuation.

I can think of 3 potential ways to improve the alias rejection:

Firstly a higher-order anti-aliasing filter to the I/Q signals before the ADC, this could be built using low-cost op amps. 

Secondly, for the cost of the additional filtering, you might as well use a high quality audio ADC with very good anti-aliasing filters built in, and far better dynamic range. 

Thirdly, reduce the bandwidth of the QSD and over-sample. For voice signals an oversampling ratio of 16x could be achieved and that would give very good rejection of aliased signals, the downside is that the frequency resolution of the PIO fractional divider would no-longer be sufficient to cover the HF bands without gaps. There are other approaches to generating the quadrature output that would give much better frequency resolution but  consume more CPU. Alternatively, an external PLL could be be used to generate the quadrature local oscillator, which would have the additional advantage of removing jitter.

I think that combining the second two approaches combining an external PLL and ADC would not only improve the rejection of alias signals but give a much performance all-round. Although I have taken a minimalist approach on the PiPicoRx, it might also be fun to experiment with some of these options in a more uncompromising and high performance receiver.

Kind Regards
Jon

w9ran, 2024-02-09:
>  Are you seeing signals of similar magnitudes at the I and Q input to the ADC?
Jon, 
Yes and no - let me explain...
I attached a header to allow monitoring of the IQ signals right at the ADC0 and ADC1 pins.  After several hours looking at the QSD output, both with o'scope and by feeding it into a soundcard on a PC running SDR software,  I'm more puzzled than ever!   The performance of the QSD seems to be frequency-dependent -- here's what I've seen:
1.  The IQ clocks going to the QSD seem to be correct, as I'd previously found, with the exception of the jitter noted.   The effect of the jitter was not immediately obvious to me as I'm still in the process of understanding how the PIO clock works.  But having captured clock waveforms in the range of 4-14 MHz the phase and amplitude at LO_1 and LO_2 are consistent in amplitude and phase.
2.  When a single tone is being received, the expected output from the QSD is the difference between the input frequency and that of the LO, with the two outputs 90 degrees apart.   For example, when tuned to 5000 kHz and a 5001 kHz tone is injected, the QSD outputs should be at 1 kHz.     In my PicoRx this is true - up to a certain frequency, but between 5 and 6 MHz, the output of the Q channel disappears.     One possible cause could be LO-related but as shown here, the 5 and 6 MHz LO clock waveforms seem identical in amplitude and phase:

[image]

This leaves the QSD itself as the remaining culprit.   I captured the QSD output  while changing the frequency the PicoRx is tuned to and the signal generator to identify where it works and where it doesn't.   Here is a 50uV signal at 5 vs 6 MHz:
[image]

As seen, the QSD is working perfectly at 5 MHz but not at 6.   This is confirmed by good image suppression (about 1.5 deg phase error) at 5 MHz vs zero suppression at 6 MHz.

3.  To further narrow down the range I worked up from 5 MHz in 100 kHz increments.   On the left, the QSD is working as it should at 5.6 MHz,  becomes erratic at 5.7 MHz and fails at 6.0 MHz:
 
[image]

4.  The input level was 50uV for these measurements.  Increasing the input to 500 uV showed an odd result.   The output that appeared to be dead showed output but one channel was badly clipped:

[image]

5.  After watching the IQ output from the QSD I now see that my earlier theory that adding capacitance was causing the LO clock to work correctly was wrong.  In fact the capacitance somehow tricked the QSD into produce output on one channel, which did bring the receiver to life, but without proper IQ input, thus the images.

6.  Just for reference, I'd note part of the secret to the high performance of Elecraft gear is their roofing filter, but being a QSD the roofing filter in the KX3 is indeed a good active filter using ADA4075-2 as you suggest.   Of course it is not a wideband front-end.

6.  I agree that your suggested changes could produce a better receiver but I like the minimalist approach and can't help but think there is something going on here that I'm missing.      Evaluation continues.
7. 
73, Bob W9RAN

dawsonjon, 2024-04-01:
I have now been able to reproduce this problem, and I think I have a fairly simple fix that works. I found that I was receiving CW signals in the SSB section of the 20m band at about 14.245 MHz. When I examined the I and Q signals on the scope, I could see that while the I channel seemed to be working, there was no output on the Q channel. Looking at the DC levels on the Q channel, I could see that although the differential inputs to the op-amp were sitting around the mid-supply point, the output was swinging to the negative rail. At lower frequencies, both I and Q channels were working and at higher frequencies neither were working. I didn;t have any success with additional termination, or reducing the drive strength on LO signals. 

I did notice that in the latest revision of the truSDX schematic, https://dl2man.de/wp-content/uploads/2023/07/truSDX_Main_1_2-1.pdf two additional capacitors had been added to the tayloe detector to remove DC bias that causes issues at higher frequencies.

I tried a similar mod on the Pi-Pico RX:
[image]

I suspect that the Pi-Pico RX was suffering from the same issue as the truSDX, because the fix seemed to completely solve the issue. I now find that the I and Q channels are both working properly across the whole frequency range. The CW signals in the SSB region of the 20M band almost completely gone. I think this also solves issue #2 which is probably another symptom of the same underlying problem.

I haven't included any IQ imbalance correction in the software yet, but this is something I plan to do at some point. I think that this should further improve the image rejection.

w9ran, 2024-04-01:
Hi Jon,
I'm glad to hear your results - after doing a lot of testing I've concluded there are at least two issues,  one of which may have been of my own creation.  Because I wanted to use my regular station antennas I omitted the preamp stage and hacked the front-end component placement to apply a Vcc/2 bias at the input to the filters, with the antenna AC coupled as usual.   I found severe degradation of the IQ signals which resulted in the ADC input (usually Q but not always) being driven to saturation, 3.2V.   When this happened the DSP failed because that channel was in essence "missing".   This only happens above about 10 MHz, on the lower bands the image suppression is quite good.   Looking at the quadrature clocks with a dual trace scope fails to show any obvious difference (as I'd expect knowing they are being generated by the PIO).  The jitter is always present but it doesn't have any effect at the lower frequencies so that can't be the reason for the IQ voltage saturation.

I took a different approach by replacing the 470 ohm resistors at the ADC inputs with 100nF capacitors.  That dropped the DC bias to about .5V so the ADC inputs were not railed, but images persisted.   I then changed my front-end jumper to bypass the filter and apply DC bias and RF directly to the QSD input (pin 9 of 3253).   At last - that resulted in good image suppression even on 20 meters!

I don't know why applying the bias before the LPFs (which have DC continuity) shouldn't work  but all the other designs I looked at apply the bias at the QSD input, after all front end filters.   So that's what I plan to do, also I think BPFs will provide greater rejection of unwanted signals that appear at the antenna input.  Another oddity:  even though the two 10K resistor divider should produce 1.6VDC, the actual bias voltage is always greater - 1.9 to 2.4VDC.  So something else is pulling this node up, but was hard to isolate the cause until I bypassed the filters.

Even so,  I still do not have proper QSD operation and/or image suppression at frequencies much above 15 MHz.   I think this may be due to a different issue - signal degradation caused by the PCB trace characteristics which become more severe as frequency increases.   As partial confirmation of this theory I found this video:  https://youtu.be/djIFIyeJyZw    It is a QSE rather than a QSD but shows how the quality of the quadrature clock signals affects the image suppression (i.e. opposite sideband suppression in his case).   This confirms what I've found on my homebrew uSDX, although I found that decreasing rather than increasing drive strength made an improvement.  Without knowing exactly what affects signal quality it's likely that various factors are in play.    As I commented, I think the routing of these HF-range signals is an important factor in obtaining optimum QSD/QSE performance, although little has been written about this to my knowledge.

Just to confirm for anyone following this - all of my comments relate to my modified implementation - I can't speak to the IQ suppression of the original design with the preamp. 

73, Bob W9RAN

w9ran, 2024-06-06:
Jon, after a diversion using my own hardware to run your Pico RX code (unsuccessfully) I'm back.   I built up a new PicoRX using your first generation PCB (no preamp) but it did not work at all on any frequency - only noise, squeaks and sqwaks even with very strong RF signals applied.   The reason evidently is the opamp outputs were saturated at the Vcc rail and cannot amplify the weak audio-frequency voltages from the sampling caps.    I'm not sure what the source of this > 3VDC voltage;  most QSDs apply a Vcc/2 DC bias to the multiplexer input which is then propagated to the sampling capacitors,  but I've encountered this problem with that approach as well.   I assume the reason is to center the RF signal within the input voltage range of the multiplexer, but I sometimes have better results by removing this bias voltage completely.   Since a voltage was present, I did a quick test by tacked a 1K resistor from the receiver side of the antenna coupling capacitor to ground - which which brought the "bias" voltage down to about 200 mV and made the receiver come alive.    Here are two videos showing without and with this resistor:

https://share.icloud.com/photos/065a_u4QCSfbUwPkxqrknpiLw

This tended to confirm the idea that somehow the opamps were being driven in to saturation, which is what your PCB change was intended to remedy.    So I proceeded to add the 10uF series capacitors in series with the 100 ohm resistors connected to the inverting inputs (R3 and R5) on the "original layout" PCB.    Even though this removed the source of unintended bias voltage, only noise could be heard - no different than the 40 sec. "before" video.  I tried a different capacitor but the result was unchanged. - the same noises but no signals could be detected even with millivolts out of the signal generator.    Finally I removed the series cap and put the 100 ohm resistor only back and as before, with the 1K resistor on the input the receiver works again.

This leaves me with more questions than even before!   My sideline approach was to make my own set of boards with the Pico,  filter, and QSD on separate PCBs so I could more easily evaluate each element of the design, but I was unable to get the Pico RX code to demodulate anything from my QSD even though it produced plenty of signals when fed into a soundcard and PC.   All of which is very frustrating!   So any information you might be able to provide would be most welcome!  For example:

- what voltage range is the ADC expecting to see from the QSD?
-  Is a DC bias expected to be present?
-  the uSDX applies a Vcc/2 bias to center the QSD signal in the ADC range - This does not seem to work with the Pico - is there any advantage to doing so?    (Again need to know the expected ADC input range)
-  What is the purpose of the 470 ohm series and 1K to ground resistors between the QSD opamp and ADC inputs?   (Perhaps voltage divider to protect the Pico input from > 3v3?)
-  Most QSDs including the uSDX use AC coupling between the op amp outputs and ADC inputs.  This would completely remove any DC bias - is there a  reason you chose not to do this?   
- Any idea why adding the capacitors in series with R3 and R5 did not work as it evidently did for you?
- A minor correction to the note on the Preamp page - to build without the preamp it is R1, nor R12, that needs to be replaced by a 0 ohm link

In my experience I've never had an SDR receiver become totally "deaf" like this one does,  which I can only assume happens when some ADC input parameters are exceeded and it simply cannot work anymore.    I'd like to use the PicoRX code with my own QSD and filter so understanding this is my #1 question.  If I can get to that point I want to proceed to compare the Pico PIO clock generation method with others (i.e. Si5351),   evaluate the image suppression across the entire HF range,   make valid MDS measurements, etc.       Thanks in advance and 73.
Bob

dawsonjon, 2024-06-08:
Hi Bob,

Good to hear from you again :-)
Here are some answers to your questions, I hope they help.

Regards
Jon

> The reason evidently is the opamp outputs were saturated at the Vcc rail and cannot amplify the weak audio-frequency voltages from the sampling caps. I'm not sure what the source of this > 3VDC [...]

I agree, this is probably the source of the issue. I take it this > 3VDC is measured at the output of the amplifier which has saturated to Vcc? What do the inputs look like? I'm expecting them both to sit around the mid-supply rail.

> Since a voltage was present, I did a quick test by tacked a 1K resistor from the receiver side of the antenna coupling capacitor to ground - which which brought the "bias" voltage down to about 200 mV and made the receiver come alive.

How did the op-amp outputs change when you did this experiment? Did the op-amps come out of saturation and start working again?

> My sideline approach was to make my own set of boards with the Pico, filter, and QSD on separate PCBs so I could more easily evaluate each element of the design, but I was unable to get the Pico RX [...]

Interesting... I would be very interested to see your results. There are a few things to think about if you are doing a side-by-side comparison:

The first is the tuning, the clock dividers in the pi-pico have quite a low-resolution, the picoRx chooses the nearest frequency which might be tens of kHz away from the dial frequency, and then applies a frequency shift to the received signal to compensate. Since the offset frequency is always different, it might be difficult to know whether the PC and the picoRx are actually tuned to the same frequency. Are you supplying the quadrature oscillator from the pico or from some external source?

If the I and Q channel were swapped, this would compound the problem and the frequency shift would have the effect of moving the signal further away.

Another possible difference is sensitivity, the ADC in the pi-pico needs at least 20dB more gain to toggle the LSB of the ADC compared to a soundcard. It is possible that the soundcard is sensitive enough to see signals that the ADC in the pico cannot. 

> what voltage range is the ADC expecting to see from the QSD?
The ADC input voltage is in the range 0 to 3.3v.

> Is a DC bias expected to be present?
Yes, ideally mid rail, but any DC is removed by software so it isn't critical. A volt or so should work.

> the uSDX applies a Vcc/2 bias to center the QSD signal in the ADC range - This does not seem to work with the Pico - is there any advantage to doing so? (Again need to know the expected ADC input range)

That is how the PicoRx is intended to work too. It doesn't need to be perfectly centred, so long as the whole signal fits inside the 0-3.3v range of the ADC it should work fine.

> What is the purpose of the 470 ohm series and 1K to ground resistors between the QSD opamp and ADC inputs? (Perhaps voltage divider to protect the Pico input from > 3v3?)

Yes, precisely. Its a potential divider to keep the (nominally) 0 - 5V swing from the op-amp outputs inside the ADC range. In an earlier prototype I didn't bother with the potential divider, I just applied a 3.3v/2 bias. It worked fine without the divider, but it didn't seem like very good practice.

> Most QSDs including the uSDX use AC coupling between the op amp outputs and ADC inputs. This would completely remove any DC bias - is there a reason you chose not to do this?

Mainly for simplicity, the ADCs need a bias so I didn't see a reason to remove it only to add it back in again. I think that this approach is needed in the uSDX because the ADC reference voltage is switched between 3.3v and 5v to give a crude form of gain adjustment. The ADC bias voltage is then changed from 3.3v/2 to 5v/2 depending on the mode to maximise headroom.

> Any idea why adding the capacitors in series with R3 and R5 did not work as it evidently did for you?

Not really, I used the approach from the truSDX design which introduced the capacitors in revision 1.2. They are briefly explained in this video at around 1:30 https://www.youtube.com/watch?v=A8Ii1UOXmtg. It sounds like it was only an issue with certain batches of the op-amp. I couldn't find any more details.

The issue I saw was that the op-amps were saturating (only at certain frequencies), sometimes to Vcc and sometimes to ground, sometimes on the I channel, sometimes Q. Both the inverting and non-inverting inputs were sitting very close to Vcc/2, and the the circuit otherwise seemed to be working correctly. Adding the capacitors to the inverting inputs seemed to fix the problem. I think that this implies that there was a minute (too small to measure with a scope, but large enough to saturate a high-gain amplifier) DC difference voltage between the inverting and non-inverting inputs that the capacitors removed.

> A minor correction to the note on the Preamp page - to build without the preamp it is R1, nor R12, that needs to be replaced by a 0 ohm link

Yes, that is a mistake, I will correct it, sorry about that. You will need replace both R1, R11 and R12 with a 0 ohm link. R12 provide a connection to the biasing network, while R1 and R11 provide a connection to the next stage in the receiver.

w9ran, 2024-06-12:
Jon - thanks for your reply and I agree, the issues seem to be related to small DC voltage offsets that are difficult to measure but which, given the extremely high gain of the amplifier stage,  result in saturation.   I understand that the Pico ADC will require more signal than the soundcard, but that should result in weaker signals, not total deafness, which results when the opamp outputs are driven to the V+ rail.      

Since the purpose of the project is learning,  I'll list the various things I tried below in case it is helpful to others:

I have a PC soundcard connected in parallel with the ADC inputs, with no observable degradation of signal.  Due to the shifting low IF of the PicoRX,  I measure the actual LO frequency and then tune the PC software to create a similar low IF offset.  The SDR software will show any images as a mirror-image on the opposite size of center and also gives a relative indication (dBFS) of image suppression.

My hardware is identical to the PicoRX except for test purposes no filters are in line, and the QSD uses a cheaper and easier to solder LM-4562 opamp, just like the uSDX uses.   However mine is running from 5V vs. 12V in the uSDX.   I am using 15nF sampling capacitors rather than 470nF as in the uSDX to obtain wide enough bandwidth for the low IF.

Test 1:  Original PicoRX PC layout with 100 ohm resistors, no capacitors in series with opamp inputs.    With antenna coupled via 100nF but no bias applied, the  sampling capacitor voltages were > 4 VDC and both I and Q opamp outputs were at the 3v3 rail.  No signals can be heard using PicoRX or PC

Test 2:   Attached the 10K bias network with a 0 ohm jumper to the receiver input.  This should have produced a DC bias of Vcc/2 or ~ 1.6V at the sample capacitors but in fact the voltage is ~ 2.09VDC.   (This was also true for the first board I built - the opamps are adding to the DC bias).   The opamp outputs are still saturated at > 3.2VDC and no signals can be heard using PicoRX or PC.

Test 3:  Connected a 1K resistor across the lower 10K of the bias divider.   This reduced the bias voltage to .39VDC at the sample capacitors and the PicoRX came to life.  Signals can be heard on both PicoRX and via soundcard, but the PC shows there is no image suppression at any frequency (and opposite sideband images are heard in the PicoRX audio).

Test 4:   All four 100 ohm resistors were removed and replaced with 100nF capacitors.    This is one of the QSD configurations that Tayloe describes in his paper but on the PicoRX hardware it does not work at all (although it does work with my hardware using LM-4562).    With these series capacitors blocking the opamp voltages, the QSD input bias voltage now measures 1.6VDC as it should.  This voltage is also present on the mux side of the sampling capacitors, but the opamp inputs all measure 4.8VDC and the otuputs are saturated > 3.2V.   No signals can be heard using PicoRX or PC.

Test 5:  The configuration of the Rev 2 PicoRX was implemented, with 100 ohm resistors on the non-inverting inputs and 100 ohms in series with 4.7uF electrolytic capacitors on the inverting inputs  (the resistors were connected to the sampling capacitors and the capacitors to the opamp inputs).  The Vcc/2 bias divider was connected, resulting in 1.72VDC on the sampling capacitors.    The resulting DC voltage at the opamp outputs (i.e. ADC inputs) is in the 120mV range, which allows the PicoRX DSP software to work as it should.   

Band conditions have been poor due to solar activity but performance is now good with no images present.   With an external powered speaker, the PicoRX "sounds like" a regular receiver with good AGC action and adequate filtering for most casual listening.

Using the PC soundcard tap I measured the MDS of the QSD using the signal analysis plug-in of SDR#.  At 7 MHz, a -120dBm signal produced 3dB SNR in 500 Hz bandwidth (this is the method used by RSGB, ARRL, and Rob Sherwood).    This is more than sufficient sensitivity to hear any signal on the band, at least in this frequency range.     More testing is planned.

Thank you very much for your efforts and especially for getting to the bottom of the very frustrating image issue! 
73, Bob W9RAN

dawsonjon, 2024-06-12:
Hi Bob,

I'm really pleased you managed to get it working and I'm looking forward to
seeing the results of your tests. I really appreciate your perseverance,
with these frustrating problems - this kind of feedback will really help to
improve the design. If you are happy that the addition of the capacitors to
the op-amp inputs is working as expected, I think that I will now re-spin
the PCB to incorporate the changes (while addressing some of the other
points you noted).

All the best
Jon

w9ran, 2024-06-13:
I measured MDS of the QSD (same as before,  3dB SNR in 500 Hz bandwidth) at 14 MHz to be -113dBm and  -103dBm at 28 MHz.   That correlates with .5 and 1.5uV which isn't bad for a super-simple receiver.   A preamp would be useful on weak signals in this frequency range.