dawsonjon/PicoRX #7: C4 and C11 - only on noninverting inputs ?

penfold42, 2024-08-09:
Thanks for publishing these project - it looks great and i look forward to building and hacking around with it.

One question - i notice C4 and C11 are only on the non inverting inputs. Why? :)

Dan Tayloe's paper has examples with 4 series caps and no caps and IIRC trusdx has none.

Error? Omission? Enhancement?

dawsonjon, 2024-08-09:
Hi,

Thanks very much for your interest in the project. There is quite a long story behind those caps...

The caps were added in response to this [issue](https://github.com/dawsonjon/PicoRX/issues/3). At higher frequencies the op-amps were saturating at the supply rail resulting in a loss of either the I or Q channel or both. This in turn resulted in loss of image rejection, or complete failure to function. The caps solve the issue which was presumably caused by a minute DC bias. It was too small to measure on a scope but enough to saturate the op-amps.

The caps were introduced to the truSDX in revision 3. They are mentioned in this video
https://dl2man.de/wp-content/uploads/2023/07/truSDX_Main_1_2-1.pdf. I think they were having the same issue, but apparently only with certain op-amps batches. I shamelessly copied the solution, credit goes to them for Manuel and Guido for finding the solution.

Thanks
Jon

penfold42, 2024-08-10:
Thanks for the quick reply !

closed 2024-08-10 by penfold42