dawsonjon/PicoRX #8: pico 2 support?

penfold42, 2024-08-10:
and will the boosted core clocks and floating point and DSP instructions help?

jumbo5566, 2024-08-20:
pico is engught,not recomm pico 2

bmentink, 2024-08-23:
Yep, request a pico 2 port please, not only is it faster 150Mhz but ADC noise problem is fixed. Any chance?
Also, how do I use the python script to change the memory table for the bands. I guess I need a python serial module?

dawsonjon, 2024-08-25:
Have added pico2 support. Quite a big performance improvement. There is a
video here: https://youtu.be/QfNhv-COuvU.
There is a crude command line application to upload memory contents from a
.csv file. https://github.com/dawsonjon/PicoRX/tree/master/memory_loader. I
think PySerial is the only dependency. You need to select "USB Memory
Upload" from the menu and then run the script. If there is interest, I
could make a portable GUI app similar to chirp.

closed 2024-08-26 by dawsonjon