dawsonjon/PicoRX #10: using a different display?

wa2mze, 2024-08-21:
I found this display, https://www.aliexpress.us/item/3256806709509463.html?ug_edm_item_id=3256806709509463&pdp_npi=4%40dis%21USD%21%246.03%21%243.50%21%21%21%21%21%4021017dd017241432938424107d1f3d%2112000038645521034%21edm%21%21%21&edm_click_module=item_detail&tracelog=rowan&rowan_id1=aeug_edm_56672_1_en_US_2024-08-20&rowan_msg_id=0220biz_search_product%3A0%3A0_575977151%240f0a6fe49d074b1588cfa4e1b5b33538&ck=in_edm_other&gatewayAdapt=glo2usa
which is larger than the one you are using.  However, it uses a different controller chip.   I wonder how difficult it would be to change the driver code for it?  Adafruit has an arduino library that should work, but you are using the SDK not Arduino.  Any ideas?

penfold42, 2024-08-22:
It claims to be compatible so if you buy the 4 pin I2C version it might just work.
The pictures show a default I2C address of 0x3c which is what this code expects

I was considering adding SH1106 support as i have one and they are 1.3" from memory, but maybe ill buy the one you linked.

dawsonjon, 2024-08-22:
Yes, I read that it was compatible, I wasn't sure if it meant electrically compatible or a drop replacement. For that price it might be worth a punt.

wa2mze, 2024-08-22:
There are two different displays that use the SH1106 and SH1116 chips, the 1.3" and 1.54"  I bought the larger one.  There is a web page out there that shows the differences between the various OLED controllers, the most important ones being in initialization (if done wrong the display won't even start or light up), and in addressing (the SH versions seem to have a max column address of 132 instead of 128 and also support only a single addressing mode).  It is possible to carve out a common feature set that will work with all of the controller versions.

MrSVCD, 2024-09-03:
Speaking of larger displays, I found these on ebay: https://www.ebay.com/itm/296119960051
SSD1309 should be compatible with SSD1306 but don't quot me on that. I only fund this on a Swedish sellers site.

bmentink, 2024-09-03:
> I found this display, [...]

I can confirm that this display works perfectly with the latest build of the code. Now my old eyes can actually read the display :)

wa2mze, 2024-09-03:
I did a git pull and loaded the new .uf2 file for the pico1 and tried the display.  It lights, but only displays random junk on the screen.  I will need to play with the OLED driver, as mine doesn't work.  Some of the Chinese displays actually have a CH1116 controller, not an SSD1309 or SSD1306.  The CH1116 internally has a 132 wide column count, not 128, so the addressing may be off if compiled for the SSD variants.  The fix may be just to init for 132 X and 64 Y instead of 128 x 64 y.

bmentink, 2024-09-03:
Yep, I noticed the same on rp2040 but it works ok on the rp2350 . .. go
figure

penfold42, 2024-09-03:
> Yep, I noticed the same on rp2040 but it works ok on the rp2350 . .. go figure. Also note, mine has CH1116 marked on it, but still works fine on the rp2350 ..

You could try some pull up resistors on SCL and SDA if the display doesn't provide them.

When it works, do you configure it as ssd1306 or sh1106 ?
The init sequences are identical. when we write the data we just offset the starting address by 2 on the sh1106

bmentink, 2024-09-03:
> You could try some pull up resistors on SCL and SDA if the display doesn't provide them.
>
> When it works, do you configure it as ssd1306 or sh1106 ? The init sequences are identical. when we write [...]

I have not touched the code at all. It runs with the default, whatever that is. I have not needed any pull-ups either, maybe the rp2350 has internal pull-ups turned on?

wa2mze, 2024-09-03:
Don't know, I haven't recompiled anything, just used the pre built uf2 file.  I don't think I need the pull ups, I can see the display being updated, and it responds to changes in frequency, etc.   Just not readable because all the pixels are in the wrong place.

bmentink, 2024-09-03:
> Don't know, I haven't recompiled anything, just used the pre built uf2 file. I don't think I need the pull ups, I can see the display being updated, and it responds to changes in frequency, etc. Just not readable because all the pixels are in the wrong place.

Do you have a pico2 you can try?

wa2mze, 2024-09-03:
Don't have a pico2 yet.  I'm on the notify list at digikey and sparkfun for one.

bmentink, 2024-09-04:
[image]
Here it is ..

penfold42, 2024-09-04:
I _definitely_ need to find a better 2X font!

bmentink, 2024-09-06:
Just discovered that the riscv build works with pico2/CH1116 display combo perfectly, but the ARM build does not, just gives screen full of random pixels ..
I.E pico2rx-riscv.uf2  works, pico2rx.uf2 doesn't ...

dawsonjon, 2024-09-07:
Have you tried these builds? https://github.com/dawsonjon/PicoRX/actions/runs/10704854534

bmentink, 2024-09-10:
> Have you tried these builds? https://github.com/dawsonjon/PicoRX/actions/runs/10704854534

Is that question aimed at me? What has changed for those builds?

ThiagoDSP, 2024-10-02:
Has anyone tested this 2.42" screen? A 2x larger screen would be great. It seems to use the ssd1309 controller

https://a.aliexpress.com/_msi84Y0

mryndzionek, 2024-10-09:
Just refreshing the thread. On `testing` branch we are using u8g2 library, so many more displays should be supported. [Here](https://github.com/dawsonjon/PicoRX/blob/testing/ui.cpp#L3057) adjusting this `Setup` function should be all that's needed. The list of available setup functions: https://github.com/olikraus/u8g2/blob/master/csrc/u8g2.h#L680 (use the ones with `_f` at end).

penfold42, 2024-11-10:
> Has anyone tested this 2.42" screen? A 2x larger screen would be great. It seems to use the ssd1309 controller
> 
> https://a.aliexpress.com/_msi84Y0

I just received https://www.aliexpress.com/item/1005006100836064.html - blue I2C in my case.
Looks the same as your link.

Works fine as a "ssd1306" in PicoRx.
[image]

It's comically large in comparison to the 1.3" SH1106 I replaced

spanglz, 2025-03-15:
Did anyone connect ST7567S display?

Like this:
https://aliexpress.ru/item/1005004252465006.html?spm=a2g2w.orderdetail.0.0.39724aa61CjZOp&sku_id=12000028529183732