dawsonjon/PicoRX #15: Features discussion

penfold42, 2024-08-29:
Is there value in a working list ?

Some ideas to start:

-   Configurable display timeout. After 'n' seconds blank and disable the charge pump. Any button or movement will first unblank the display

-  Adjustable RF gain. If you add a PE4302/PE4312 attenuator  to the 13dB preamp you could set the RF gain from -18.5 to +13dB gain. Ideally we'd use SPI bus to control it

dawsonjon, 2024-08-31:
I like the ideas, keep them coming. I'm working on a branch at the moment that has a configurable display timeout. Also working on improving the filters and doing more of the DSP in the frequency domain.

mryndzionek, 2024-09-07:
Minor improvements:
 - exit the menu just one level up instead of going to the main screen while pressing `back`

Bigger functionalities:
 - SD card recording

dawsonjon, 2024-09-07:
Sounds good to me!

mryndzionek, 2024-09-08:
One more small feature might be encoder sensitivity setting. I've used an encoder with twice the detents than pulses and so need to have:
```patch
--- a/ui.cpp
+++ b/ui.cpp
@@ -24,7 +24,7 @@ void ui::setup_encoder()
 
 int32_t ui::get_encoder_change()
 {
-    new_position = -((quadrature_encoder_get_count(pio, sm) + 2)/4);
+    new_position = -((quadrature_encoder_get_count(pio, sm) + 1)/2);
     int32_t delta = new_position - old_position;
     old_position = new_position;
     if((settings[idx_hw_setup] >> flag_reverse_encoder) & 1)
```

mryndzionek, 2024-09-13:
Two more suggestions:
 - signal strength display on the `Recall` screen
 - scanner and scanning mode

penfold42, 2024-09-14:
[image]

I've added signal strength to Recall screen but cant decide how to present it.
- Inverted text somewhere (at the top in the example)
- Sneak a bar in somewhere (there are 3 in the example)

If I had to pick, I'd go for the inverted text at the top and the bar at the bottom...

mryndzionek, 2024-09-14:
> ![recall_ss](https://private-user-images.githubusercontent.com/13173376/367476698-822a31ee-6d9a-4b82-9bc5-4b42b568efe9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYyOTQ0MDQsIm5iZiI6MTcyNjI5NDEwNCwicGF0aCI6Ii8xMzE3MzM3Ni8zNjc0NzY2OTgtODIyYTMxZWUtNmQ5YS00YjgyLTliYzUtNGI0MmI1NjhlZmU5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE0VDA2MDgyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNmNWU1ZTgzNjg1NDM1ZTAyMWU3YTE0MDk0YWUwZTM2ODQxNWE1NjQyNGRiMmYxYjk3MTNjNWM2NWQ4MTFjNWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.xRNH--DWHOGeC6glsG5zCvCcfznjMufVOA0UR15rAh8)
> 
> I've added signal strength to Recall screen but cant decide how to present it.
> 
>     * Inverted text somewhere (at the top in the example)
> 
>     * Sneak a bar in somewhere (there are 3 in the example)
> 
> 
> If I had to pick, I'd go for the inverted text at the top and the bar at the bottom...

Maybe a thin vertical bar?

penfold42, 2024-09-14:
I think you win....there was just enough room to squeeze it in.
I've just pushed https://github.com/penfold42/PicoRX/tree/recall_SS