dawsonjon/PicoRX #19: Configurable amp/preamp gain? So S meter is more accurate

penfold42, 2024-08-31:
Do you want me to add it?

With breadboard and original versions around plus whatever random variations there are in the wild I think it has value.

If so, should I remove the  preamplifier_gain_dB and just make amplifier_gain_dB dynamic?
I notice both get used in get_signal_strength_dBm() but only amplifier_gain_dB is used to compute the s9_threshold

dawsonjon, 2024-08-31:
Yes, I agree. I think it makes sense to make it as configurable as possible without the need to rebuild. I'm thinking that it might be worth adding a sub-menu for hardware configuration type items? For now, I think it makes sense to have a single "system gain" value as you suggest, it could then also be used to "calibrate" the S-meter. I think the s9_threshold thing was probably an oversight rather than being intentional.

penfold42, 2024-08-31:
+1 on the submenu. 

I'll do the system gain item next

penfold42, 2024-08-31:
Mostly done I think - https://github.com/penfold42/PicoRX/tree/gain
Except for a sensible compile time default.

Where/How are the default config settings applied at build time?

dawsonjon, 2024-08-31:
Great! I think its just a matter of initialising that sector of flash in the header. Each row is a copy of the "settings" array, the first row determines the default setting.
https://github.com/dawsonjon/PicoRX/blob/master/autosave_memory.h

penfold42, 2024-09-01:
#23

closed 2024-09-01 by penfold42