dawsonjon/PicoRX #20: double size fonts in menus?

penfold42, 2024-08-31:
I've started playing around with this in my  [testing branch](https://github.com/penfold42/PicoRX/tree/testing)

I've had to reword the longer menu items to fit on one line and I've also reworked the Frequency menu test entry. As part of that, the oled library can now draw in black or white.

Thoughts? 

If you think its a good idea, do we make it configurable ?

dawsonjon, 2024-08-31:
Sounds good... the display is rather small. Will check out your branch when
I get a moment.

dawsonjon, 2024-09-02:
I had a look at your branch, and it looks really good! I love what you did with the frequency entry! I wonder if we should re-work the menu to split the longer items over multiple lines? I think we could then have a larger font without having to abbreviate the names too much?

penfold42, 2024-09-02:
:) that damned "colour" variable had to be added to EVERYTHING.. just for some poxy reverse video!

Maybe ill make the string print commands just wrap text that doesnt fit. that way if we add bigger screens later we have less rework ? And i guess support and explicit "\n" too
Also makes it easier if we want to have the menu font size configurable.

What do we do with the recall option? ditch some of the details if we're in 2X ?
save is is easier cos its only a name which we can wrap.

dawsonjon, 2024-09-02:
I was thinking that with word wrapping, we would be able to make a good compromise between readability and meaningful names that might be able to make a permanent change rather than an option.

I had forgotten about the recall option!

penfold42, 2024-09-03:
Check out #32

closed 2024-09-05 by penfold42