dawsonjon/PicoRX #31: Dithered waterfall

mryndzionek, 2024-09-01:
I'm playing with dithered waterfall. First tests:
[image]
What do you think? Is it useful?

mryndzionek, 2024-09-01:
Or maybe something like this:
[image]

dawsonjon, 2024-09-01:
That's very cool! Love it! How did you generate the .gifs?

mryndzionek, 2024-09-01:
> That's very cool! Love it! How did you generate the .gifs?

Pushed the framebuffer via UART and decoded in Python.

mryndzionek, 2024-09-01:
Here is a gif with adjusted framerate:
[image]

dawsonjon, 2024-09-01:
That looks even better!! How about adding a black pixel or two above each bar in the spectrum scope, so that the spectrum scope is more distinct from the waterfall? Might be better for strong signals where the waterfall will be mostly white?

mryndzionek, 2024-09-01:
> That looks even better!! How about adding a black pixel or two above each bar in the spectrum scope, so that the spectrum scope is more distinct from the waterfall? Might be better for strong signals where the waterfall will be mostly white?

[image]

penfold42, 2024-09-01:
very cute! 

Have you optimised the ssd1306_show function?
Would it help? A full update is 132*8=1056 bytes,  at about 40 kByte/sec = 26ms
A simple 'which pages are dirty' should get some good gains.

mryndzionek, 2024-09-02:
> very cute!
> 
> Have you optimised the ssd1306_show function? Would it help? A full update is 132*8=1056 bytes, at about 40 kByte/sec = 26ms A simple 'which pages are dirty' should get some good gains.

No, no optimizations in thin regard were done. I'm not sure it's worth it. Most pages will be 'dirty' every 100ms, no?

penfold42, 2024-09-02:
The top 3 pages(of 8 vertical pixels) with frequency and voltage etc are essentially static.
And as best I can tell, only the middle ~64 pixels are used in the spectrum.

Speaking of which - how much spectrum can we paint? at a guess its currently +/- 3kHz

dawsonjon, 2024-09-02:
In the current version the whole spectrum is 12.5kHz, but is captured after the filters leaving 6.125kHz. I'm working on a version in the testing branch that does the filtering in the frequency domain, with a fixed decimation of 16, this should give 31.25kHz of spectrum.

dawsonjon, 2024-09-02:
Just spotted the last gif. Looking great!

mryndzionek, 2024-09-02:
> The top 3 pages(of 8 vertical pixels) with frequency and voltage etc are essentially static. And as best I can tell, only the middle ~64 pixels are used in the spectrum.

The waterfall "bitmap" height is 29 in my screenshots.

mryndzionek, 2024-09-02:
Regarding UI changes, I was thinking about putting most of the "metrics" at the top with even smaller font, making the S-meter thinner with possibly just a segmented bar. Here is a prototype:
[image]

dawsonjon, 2024-09-02:
Looks good, I like the concept. Making the s-meter.more compact/nicer is a great idea. Using a battery icon saves plenty of display real-estate and looks nicer too, the level of 'fullness' depends on the chemistry/number of cells, so some configuration might be needed. Also think it should be possible to detect whether USB power is present and modify icon accordingly. 'lightening symbol'? It might be nice to find some way of representing the filter bandwidth too, perhaps a simple horizontal bar to represent the pass band? There are probably some compromises that could be made, I know the display is a bit cluttered and it might be reasonable to leave off things like CPU temp and CPU load, but I figured it was likely to be used by people with a technical background who might like to know all the facts and figures.

penfold42, 2024-09-02:
Maybe move the hardware monitoring to a separate screen ?
You dont need it when listening to some phat beats on broadcast AM or ringway manchester calling CQ for a YT vid

penfold42, 2024-09-02:
> > The top 3 pages(of 8 vertical pixels) with frequency and voltage etc are essentially static. And as best I can tell, only the middle ~64 pixels are used in the spectrum.
> 
> The waterfall "bitmap" height is 29 in my screenshots.

I meant the width - we dont have to write all 128 bytes horizontally

mryndzionek, 2024-09-02:
> > > The top 3 pages(of 8 vertical pixels) with frequency and voltage etc are essentially static. And as best I can tell, only the middle ~64 pixels are used in the spectrum.
> > 
> > 
> > The waterfall "bitmap" height is 29 in my screenshots.
> 
> I meant the width - we dont have to write all 128 bytes horizontally

Okay, sorry. The `spectrum` array is currently 128 bytes long, so I went with it.

bmentink, 2024-09-04:
> Regarding UI changes, I was thinking about putting most of the "metrics" at the top with even smaller font, making the S-meter thinner with possibly just a segmented bar. Here is a prototype: ![pico_rx_ui](https://private-user-images.githubusercontent.com/786191/363599069-54da889d-0df2-4748-b64c-7e2eb32c179a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU0MTA3MDQsIm5iZiI6MTcyNTQxMDQwNCwicGF0aCI6Ii83ODYxOTEvMzYzNTk5MDY5LTU0ZGE4ODlkLTBkZjItNDc0OC1iNjRjLTdlMmViMzJjMTc5YS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwNFQwMDQwMDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMjYyZjBhNWQ5NmU0ZmY4ZDllNDEyN2U3MGQ4NTBmZTg2ZDhiNTc0OTFkMzZkOTY1Yjg2NjEyYjA1NmI4ZjI0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ZH0GL9TbKUhcm890JT2jXvsD0LzicTNxghMO3swRFhQ)

Like it!
Could you push it to a branch? (Would look really good on the bigger 1.5" display as discussed on other thread)

penfold42, 2024-09-04:
On a related topic, the 2X font might benefit from being a "native" 10x16 font rather than a 2x scaled 5x8

mryndzionek, 2024-09-04:
> Like it! Could you push it to a branch? (Would look really good on the bigger 1.5" display as discussed on other thread)

The prototype was created using Inkscape. I made some attempts to recreate it in u8g2, but it seems too ambitious, i.e. 128x64 is a too low of a resolution to implement it. I can't make the top look similar to my prototype without significantly reducing the height of the FFT display, which is not a good thing.

penfold42, 2024-09-04:
> The prototype was created using Inkscape. I made some attempts to recreate it in u8g2, but it seems too ambitious, i.e. 128x64 is a too low of a resolution to implement it. 

The best we can can do is a x5 font. Not sure the 3px win for the waterfall is worth it! :)

mryndzionek, 2024-09-04:
> > The prototype was created using Inkscape. I made some attempts to recreate it in u8g2, but it seems too ambitious, i.e. 128x64 is a too low of a resolution to implement it.
> 
> The best we can can do is a x5 font. Not sure the 3px win for the waterfall is worth it! :)

The main question is, do we want to integrate u8g2 (are there any licensing problems, etc)? I wouldn't attempt revamping the UI without it. The fonts support in u8g2 is fantastic: https://github.com/olikraus/u8g2/wiki/fntlistall

mryndzionek, 2024-09-04:
> The main question is, do we want to integrate u8g2 (are there any licensing problems, etc)? I wouldn't attempt revamping the UI without it. The fonts support in u8g2 is fantastic: [...]

Here is the prototype implemented in u8g2:
[image]

dawsonjon, 2024-09-04:
I think this has a lot of potential, I would be tempted to take the best elements from this design and incorporate them into the existing UI. The waterfall is very cool, I think combining this with a graphical s-meter and battery monitor would certainly elevate the UI.

U8g2 provides a neat solution. I don't think any licensing issues would be insurmountable. Would it add much overhead in terms of memory compared to the existing library? I think I would be in favour of adopting it if it does what we need.

mryndzionek, 2024-09-04:
> Would it add much overhead in terms of memory compared to the existing library? 

No, I don't think so. It's design to only compile-in the stuff that is used and regarding RAM use it also supports page-by-page drawing, so even less RAM is theoretically needed.

mryndzionek, 2024-09-05:
I'm attaching a test UF2 file with the u8g2 screen and only the frequency adjustment working. It has different IO handling - more reactive, without any inline blocking. The max. lag is then very deterministic and not bigger than ~5ms (single display page write) and no possibility of events being lost.
[picorx_oled_test.uf2.zip](https://github.com/user-attachments/files/16889025/picorx_oled_test.uf2.zip)

bmentink, 2024-09-05:
Looks good. Except, I have some rubbish characters left column of the screen ..
[image]

mryndzionek, 2024-09-05:
> Looks good. Except, I have some rubbish characters left column of the screen .. Sorry, could not eliminate the scan lines ..
> 
> ![IMG_20240906_100403461](https://private-user-images.githubusercontent.com/10123911/364967922-46442305-3fb4-4a3c-8652-adca4520e13b.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU1NzUwNDIsIm5iZiI6MTcyNTU3NDc0MiwicGF0aCI6Ii8xMDEyMzkxMS8zNjQ5Njc5MjItNDY0NDIzMDUtM2ZiNC00YTNjLTg2NTItYWRjYTQ1MjBlMTNiLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTA1VDIyMTkwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA2MmMyOWY5NjExNGM0ZWU4MTkzNTM3NmM4YWMzODcxNWI4MjVjZGUxYjI5MGIzNTY5ODIxYTZiNzNiYTE2YzgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.VsGmHpWAD0NJ7nlmwvM3yxnJIVTve0h_dZ-7wlplk5o)

Is this a SSD1306, or SH1106 display? Probably wrong controller. I think the binary was for SH1106.

bmentink, 2024-09-05:
Ahh, it was the 1306, I wasn't sure. I presume the code is for rp2040? I will try it on my ch1116

mryndzionek, 2024-09-05:
> I presume the code is for rp2040?

Yes, RP2040.

dawsonjon, 2024-09-06:
Like the new look!