dawsonjon/PicoRX #36: PicoRX Breadboard values and questions.

MrSVCD, 2024-09-04:
I am planing to make a PicoRX-breadboard and I have a couple of questions:

[image]
Is this the correct values for the color codes?

Is the audio out stage correct? It is different from the schematic.

Can I use fst3253mtcx instead of 74CBTLV3253? [https://www.electrokit.com/en/fst3253mtcx-wtssop-16-dual-1-of-4-multiplexer](https://www.electrokit.com/en/fst3253mtcx-wtssop-16-dual-1-of-4-multiplexer)

dawsonjon, 2024-09-04:
> Is this the correct values for the color codes?
Yes

> Is the audio out stage correct? It is different from the schematic.
It should work either way round.

> Can I use fst3253mtcx instead of 74CBTLV3253? https://www.electrokit.com/en/fst3253mtcx-wtssop-16-dual-1-of-4-multiplexer
Yes, but its a 5v device so needs to be powered from 5v rather than 3.3v. 

Best of Luck
Jon

penfold42, 2024-09-04:
There is 1 wire missing from the schematic - the rotary encoder's push button goes to GPIO 5

MrSVCD, 2024-09-05:
Can I replace 56nF to 68nF?

dawsonjon, 2024-09-05:
Yes, close enough👍

MrSVCD, 2024-09-05:
Parts have been ordered, I have a deadline of October 1 so hopefully I can get it built before then. https://arcticradio.net/schedule

MrSVCD, 2024-09-12:
Progress report on my build.
[image]
[image]

dawsonjon, 2024-09-12:
Nice! 👍

MrSVCD, 2024-09-13:
What does the buttons do?
Backstory: I am reusing a old slide image viewer as the base for my control panel and my rotary switch push button is quite stiff so the viewer moves around if i mount it on the side.

MrSVCD, 2024-09-13:
I now have a working radio! It is late here so I will test it more tomorrow.

penfold42, 2024-09-14:
> What does the buttons do? Backstory: I am reusing a old slide image viewer as the base for my control panel and my rotary switch push button is quite stiff so the viewer moves around if i mount it on the side.

The encoder push button is not (presently) required.
- in the home screen it jumps straight into Recall
- in the menus, doubles as the right (MENU) button

MrSVCD, 2024-09-14:
[image]
Accidental power supply, did not think it would work.

[image]
Complete radio.

[image]
So the next goal is to build the radio fully into this viewer since there is quite some space in it.

[image]
[image]

penfold42, 2024-09-15:
I love it!

I may have to get some toothpicks, blutak and 3 credit card size fresnel magnifiers....

MrSVCD, 2024-09-23:
I am now satisfied with my build. I received my first LW station so I very happy.

closed 2024-09-23 by MrSVCD

dawsonjon, 2024-09-23:
Cool, really pleased you got it working!