dawsonjon/PicoRX #41: Performance improvements suggestions

mryndzionek, 2024-09-07:
I've finally received the last components and build Tayloe detector. Assembled everything partisan style on a veroboard. It definitely works. Here is a sample recording:
[pico_rx_1.mp3.zip](https://github.com/user-attachments/files/16914856/pico_rx_1.mp3.zip)

There seems to be however a lot of noise everywhere. What can I do to improve this? Bandpass filters? Better antenna? Maybe some wires need to be shielded (LO_Q, LO_I?)?

penfold42, 2024-09-07:
Mines pretty crappy too but i've used even worse components:
- 74HC4051 and MCP6002 (rather than MCP6022).
- no preamp at all. But I have some high power AM broadcast towers nearby
- It might also have developed a defect. Swapping IQ no longer makes any difference!

Some possible causes of noise:
- Grounding is important - vero/donut/breadboard has no ground plane so its tricky to keep the noise down.
- separate analog/RF and digital VCC and GND connections.
- You might have some LO radiating into the amp if they're nearby.
- Do you have an LC band filter? My hack job has some provision for dropping these in.
- OLED charge pump noise. The TruSDX guys were modifying the oled so they could disable it IIRC

Tonytti, 2025-01-17:
To use band pass filter is a must. I build pico rx and i use an external preselector/band pass filter for any ham band. It works well, apart 40 meter at night due a strong Chinese broadcast station at about 7.250 who produces several intermodulation products, it's strange because tayloe detector should be a "bomb proof" mixer. Maybe LO is not very clean? I will investigate.

mryndzionek, 2025-01-17:
> an external preselector/band pass filter

Any details on this? DIY, or bought? I added a simple 30MHz low-pass which improved the situation a little bit. I also was experimenting with a very simple hand-tuned preselector, but it was too inconvenient to tune.

penfold42, 2025-01-17:
Could you breadboard some of the filters from the [PCB version](https://github.com/dawsonjon/PicoRX/blob/master/PCB/pico_rx/pipicorx.pdf) ?

dawsonjon, 2025-01-17:
@Tonytti 

> It works well, apart 40 meter at night due a strong Chinese broadcast station at about 7.250 who produces several intermodulation products, it's strange because tayloe detector should be a "bomb proof" mixer. Maybe LO is not very clean?

Given that the interfering signal is within a few hundred kHz, I think what you are seeing here is ADC aliasing rather than intermodulation. if you have exact frequencies I can confirm, but there have been other reports. The Tayloe detector rolls off at about 12dB per decade, so a strong signal outside the Nyquist zone of the ADC might still get through. There is a very simple fix that can greatly improve the alias rejection, and that is to add a simple RC low pass filter between the op-amp and ADC. A 5.6k series resistor followed by a 1n capacitor (repeat this for the I and Q ADC inputs) works well.

As you say, the Tayloe detector is very robust to overloading and is mostly only sensitive to interfering signals at odd harmonics of the LO. For example, if you are working the 40m band, you might also see signals around 21MHz. That's where a preselector or band pass or low pass filter might help.

(Of course if you have an LNA, it will likely have a much lower IP3 than the Tayloe detector and you might see some IMD).

Tonytti, 2025-01-17:
Thanks a lot for your answer. I'll let you know...

Tonytti, 2025-01-17:
@mryndzionek 
Adding a 30 Mhz low pass filter doesn't help. The simplest way is mounting lpf just as indicated on the reported schematic on git hub. For more severe filtering band pass filters are suggested for any band of interest. It is particularly important on ham bands where generally you deal with weak signals compared to the broadcast stations ones.

Tonytti, 2025-03-03:
...i bypassed all connections to oled display and now it is quite silent. An impedence and bog capacitor on 3.3 volt. Other two impedances on scl and sda. Now it's good

dawsonjon, 2025-03-03:
Cool, that's great 👍 I would be interested to know the details, what component value did you use?