dawsonjon/PicoRX #44: Allocate GPIOs for SPI/SDcard interface?

penfold42, 2024-09-07:
SPI1 on GPIO10,11,12,13 ?
and maybe GPIO 14 and 15 for additional SPI slave selects (or multiplexing via a HC139 ?)

dawsonjon, 2024-09-07:
I have been working on a ili9341 TFT waterfall, I've provisionally allocated allocated these pins. 

    #define PIN_MISO 12
    #define PIN_CS   13
    #define PIN_SCK  14
    #define PIN_MOSI 15 
    #define PIN_DC   11
    #define PIN_RST  10

Obviously, I'm not actually using the MISO pin, but thought it made sense to allocate it so that the SPI could be used with other peripherals such as SD card or similar. I will ditch the RST pin if I can, and might try to double up the DC pin with something else too if I can.  I would like to keep GPIO28 spare for a microphone input, and 3 other pins for TX functions. Perhaps we should allocate GPIO9 as SD card CS pin?

MrSVCD, 2024-09-09:
Since you are asking, what are your plans for the SD card?

penfold42, 2024-09-09:
@mryndzionek suggested sd card recording in the #15 features discussion issue

@dawsonjon re: PIN_CS on gpio 13 vs 9. Not sure there's much benefit one way or the other.
Are we rapidly running out of pins? :) I guess theres 3 options:
- a pico2 board with the bigger package
- layout a board with the bigger package on it
- use a I2C gpio expander for the slower speed stuff (buttons, but not encoder)

MrSVCD, 2024-09-13:
If you go the I2C expander route, you could "just" use another Pico. It is a bit more expensive than a MCP23017 but gives more options.

penfold42, 2024-09-14:
> If you go the I2C expander route, you could "just" use another Pico. It is a bit more expensive than a MCP23017 but gives more options.

Quad core radio :)

Beware the I2C MCP23017 - silicon bugs require A7 and B7 be set to outputs

mryndzionek, 2024-09-14:
What about the bandpass filter control pins (2, 3, 4)? The breadboard version doesn't use those pins.

penfold42, 2024-09-14:
Ideally the one firmware would work on both with no runtime config. 

For best image rejection (and noise) you should really use the band select filters.

MrSVCD, 2024-09-15:
As I was looking for other ways to add buttons. Can ADC2 be used to add a keypad?
There is a way to add a keypad using a bunch of resistors.
Here is some maths and designs [Keypad.zip](https://github.com/user-attachments/files/17006558/Keypad.zip)
The only change to the circuit when the keyboard is not installed would be a 10k resistor from ADC2 to ground. The RP2350 has a currently a bug for the internal pull-down resistor and I don't know the value so a external one would be a safer bet.

dawsonjon, 2024-09-15:
I'm quite keen to keep the spare ADC input as a mic input when I add TX capability. It will probably double as a morse paddle input too. 

Another option to consider is to add an IR remote. Only needs one IO pin... IR remotes are also very cheap, often cheaper than a keypad.

MrSVCD, 2024-09-19:
Since I am computer collector I have some Apple Desktop Bus stuff and it uses only one I/O pin. There is another signal in the standard that is only used for a remote power button.