dawsonjon/PicoRX #65: SSD1306 does not work after Pico power off

Crouze, 2024-09-14:
Hi,

I am experiencing odd behaviour with the Pico2 and SSD1306 display which may either be "normal" (I hope not) or weird.

I have a Pico2 with just the SSD1306 connected to it, I can see it on my Linux machine with BOOTSEL, and drag and drop the pico2rx.uf2 file to it. The Pico2 then reboots as it should and the display works. But, when I unplug the Pico2 from the USB cable and re-connect it, the display stays off.

I have checked this with two Pico2's, and two SSD1306 displays. I have also checked it with both the precompiled binary and a self compiled uf2 from the master branch.

Flashing the Pico2 with blink.uf2 as a test shows that the binary does survive a power off and on.

Checking with picotool (sudo picotool info -a) shows that the pico2rx binary is actually still on the Pico2 after a power off and on.

I went so far as building the full PicoRX on a breadboard, not expecting much or any reception but at least to prove the radio is working, and indeed after a power off and on the display is "dead" but I can still hear the radio working, the noise changing with every click on the rotary decoder, and (I believe) the AGC kicking in after a few seconds.

I am at a loss what is going on here, I have excluded faulty hardware (Pico2 or SSD1306) and radio build as the behaviour is present with only the SSD1306 connected to the Pico2. Any ideas, anyone?

Thanks,
Marco

penfold42, 2024-09-14:
Some other reports on #10 of weird display vs pico vs pico2 issues.

Are you using arm or riscv ? https://github.com/dawsonjon/PicoRX/actions/runs/10760618022
Try the other :)

Could you upload a photo of the back of the OLED display?

Crouze, 2024-09-14:
Ah!

Thank you for the quick reply.  I had only been trying arm thus far, not tried riscv and it immediately works!

See the attached photos of the display.

[image]
[image]