dawsonjon/PicoRX #71: testing: new font bugs/changes

penfold42, 2024-09-16:
I love the new font!

I've found a couple of issues and i expect there to me more.

- [ ]  You broke my animated splash screen! :)

- [ ]  lower case f, s and t are offset to the left (already got a fix for it)

- [ ] the 12 wide font specified no gap between chars, the 5 wide font specifies 1px gap. I suspect some of my code like centering might get a bit confused

I'm happy to cover all 3 above. In addition, as part of the scanner work I've been secretly doing, I added another 6 characters to the font. I'll also add them to the big one.

If anyone find other UI weirdness that might be due to the fonts, add em here

mryndzionek, 2024-09-16:
Probably unrelated bugs I noticed:
 - [ ]  in the waterfall screen there is some weirdness at the top - some pixels are set above the tick marks
 - [ ] settings changes are not applied immediately after exiting (specific) settings screen, but only after some event, like encoder change

penfold42, 2024-09-16:
> Probably unrelated bugs I noticed:
> 
>     * [ ]   in the waterfall screen there is some weirdness at the top - some pixels are set above the tick marks

is that the commas rather than decimal points between? other than those I cant see it on mine.

I did also notice that the tick marks on the spectrum view are broken tho!

mryndzionek, 2024-09-16:
> is that the commas rather than decimal points between? other than those I cant see it on mine.

Yes, so there are commas instead of dots, but also it seems that scrolling starts too high and the commas are "stretched" down. You can see it happening immediately after switching to the waterfall page.

penfold42, 2024-09-16:
> > is that the commas rather than decimal points between? other than those I cant see it on mine.
> 
> Yes, so there are commas instead of dots, but also it seems that scrolling starts too high and the commas are "stretched" down. You can see it happening immediately after switching to the waterfall page.

With the aid of my phone zoom, this should be fixed in #72