dawsonjon/PicoRX #83: Audio cuts out every ~8s (every suspend cycle)

mryndzionek, 2024-09-19:
Very noticeable. Probably due to the ramping down and up of PWM. Something like this might help:
```patch
diff --git a/rx.cpp b/rx.cpp
index 15f3640..f48c2af 100644
--- a/rx.cpp
+++ b/rx.cpp
@@ -334,8 +334,11 @@ void rx::run()
 {
     while(true)
     {
-      apply_settings();
-      pwm_ramp_up();
+      if (settings_changed)
+      {
+        apply_settings();
+        pwm_ramp_up();
+      }
 
       //read other adc channels when streaming is not running
       uint32_t timeout = 1000;
@@ -373,8 +376,11 @@ void rx::run()
             adc_set_round_robin(0);
             adc_fifo_setup(false, false, 1, false, false);
 
-            //slowly ramp down PWM to avoid pops
-            pwm_ramp_down();
+            if (settings_changed)
+            {
+              // slowly ramp down PWM to avoid pops
+              pwm_ramp_down();
+            }
 
             break;
           }

```

penfold42, 2024-09-19:
+1 on the problem and that solution seems to work for me too

mryndzionek, 2024-09-19:
Pushed as #86

closed 2024-09-25 by mryndzionek