dawsonjon/PicoRX #92: NE5532P opamp issue

mryndzionek, 2024-09-25:
I've tested NE5532P opamp. It's 10MHz and noise density of 5 (at 1kHz), but it doesn't seem to be working. Recommended minimum supply voltage is 5V. I'm powering from USB, which after diode drop is 4.6V. Could this be the issue?

dawsonjon, 2024-09-25:
I haven't tried it myself, but it's quite a popular option for this type of thing.  You could try a 5v power supply to eliminate the voltage drop as a possibility? Maybe power on Vbus rather than Vsys? What are the symptoms? Is the op-amp saturating or sitting around mid supply rail?

mryndzionek, 2024-09-25:
> I haven't tried it myself, but it's quite a popular option for this type of thing. You could try a 5v power supply to eliminate the voltage drop as a possibility? Maybe power on Vbus rather than Vsys? What are the symptoms? Is the op-amp saturating or sitting around mid supply rail?

The receiver shows extremely low signal, barely breaking S0 squelch on some frequencies. I also tried LM358P and it works! Just a little bit more noise than MCP6022. I think this is faulty chip. I'll have some more in few days, so I'll check then.

closed 2024-09-26 by mryndzionek

kf4mot, 2025-12-30:
> I think this is faulty chip. I'll have some more in few days, so I'll check then.

Just curious if you have any follow up on this? Or your like me and half your new parts get stuffed away or buried under a pile :)