dawsonjon/PicoRX #93: First weather fax decode !

mryndzionek, 2024-09-26:
Pinneberg at 3855kHz. Not bad for urban environment and indoor DIY YouLoop antenna:
[image]

mryndzionek, 2024-09-26:
The pattern of short horizontal lines is due to suspend via timeout.

closed 2024-09-28 by mryndzionek

dawsonjon, 2024-09-28:
Those are very impressive results! I think it might be worth adding a facility to disable the battery monitor for digital modes, and we could probably reduce the frequency of the updates so that they are less intrusive.

mryndzionek, 2024-09-28:
> Those are very impressive results! I think it might be worth adding a facility to disable the battery monitor for digital modes, and we could probably reduce the frequency of the updates so that they are less intrusive.

We are suspending just to read the voltage and temp., right? So actually it's not so bad to only have those values updated when the settings change.

dawsonjon, 2024-09-28:
Yes, agreed. There might be periods when the same station is being to and the battery voltage wouldn't update, but I wouldn't have thought it would be necessary to check more than once every minute or so.