dawsonjon/PicoRX #102: Distortion question

mryndzionek, 2024-09-30:
I'm attaching a sample wav:
[distortion.wav.zip](https://github.com/user-attachments/files/17190594/distortion.wav.zip)

There is this strange distortion, like wind blowing on a microphone. Is this a sign of something getting overloaded?

dawsonjon, 2024-09-30:
Does it only happen with very strong signals? I haven't allowed for worst case growth in the FFT, but instead chose a scaling that seemed to avoid overflow even for strong stations (that saturate the op-amps). It's possible that there might still be some overflow with very strong stations. To check, you could try setting extra-bits to zero, this should avoid overflow even for worst case scenario.

dawsonjon, 2024-09-30:
> AGC - not much difference, extra-bits also. It happens on fairly strong
> signals (S7-S9), but seems to randomly go away. It's also on frequencies
> above 10-13MHz. Sometimes it sounds like a different voice "trying" to come
> through. I wonder if it's some FM 88-108 interference.

Fair enough, I tend to find that interference from the oled display tends
to worsen at higher frequencies, does it make any difference when the
display is disabled? It couldn't hurt to try a broadcast FM trap if you
have one, the Tayloe Detector is sensitive to interference at odd
harmonics, and the 7th/9th harmonic would sit in the broadcast FM band at
this frequency range.

mryndzionek, 2024-09-30:
> Fair enough, I tend to find that interference from the oled display tends to worsen at higher frequencies, does it make any difference when the display is disabled? It couldn't hurt to try a [...]

Now it has gone away and the sound is very good. Actually I am using a FM trap, but DIY and unmeasured, so I don't know if it works :)

penfold42, 2024-10-02:
> Now it has gone away and the sound is very good. Actually I am using a FM trap, but DIY and unmeasured, so I don't know if it works :)

Was that after my PR which disabled I2C writes if the display is off?

Do you have the band LPF on the front end ?

mryndzionek, 2024-10-02:
> Was that after my PR which disabled I2C writes if the display is off?

It was before. At night some stations sound perfect. I have some strange local interference. I also put my LNA into a metal box.

> Do you have the band LPF on the front end ?

No, no LPF. Just a DIY FM trap.

penfold42, 2024-10-02:
Many AM transmitters lower their power at night - maybe try an attenuator during the day to see if that helps?
Which reminds me of one of my todo items - control an external attenuator like a pe4302.

I wonder how hard it would be to implement overload detection. At the ADC its easy - count # of samples that are too close to 0 or 3.3v but I'd like to detect clipping that could be introduced by the opamp too...
This would become super useful if i play with an external attenuator to to AGC on an LNA