dawsonjon/PicoRX #99: Show us your build !

penfold42, 2024-09-30:
No matter how modern, steampunked or deviated from the recommended design :)

I'll kick things off...
Mine has a few substitutions based on what was to hand:
TL072 for the opamp
74HC4052 scavenged from a Pace set top box :)
The four tayloe caps are 27nF cos the 74HC4052 has a higher Ron
Small proto area that I can use for band low pass filters (the green wire.
No preamp.

Its good for about 10MHz, FT8 on 14MHz is buried in noise.
Local AM stations are fine on an old skool wire loop antenna and some will totally overload the radio when on my EFHW.
The EFHW works well for 40m stuff.

Top view with SH1106 and buttons - the 2 boards could be folded back to back
[image]

Close up of the RF front end board
[image]

And some pretty average soldering...
[image]

dawsonjon, 2024-09-30:
Brilliant 👍 I love this stuff 😁

mryndzionek, 2024-09-30:
My first build was similar, but I fried some components while experimenting with random wire antenna and grounding. The second one is a little bit tidier :smile:

penfold42, 2024-09-30:
> My first build was similar, but I fried some components while experimenting with random wire antenna and grounding. The second one is a little bit tidier 😄

pictures or yours doesn't exist...

you dissen my neatness bro??
When I get around to ordering, ill probably just replace the RF board with the 3253 and an smd 6022 both on the green breakout.
At least it'll be mostly 2 dimensional

mryndzionek, 2024-09-30:
This is my current build. I run out of toner transfer paper, so had to draw it using a sharpie :face_exhaling: 
[image]
[image]
[image]

dawsonjon, 2024-09-30:
Nice! Love the hand etched PCB 👍

MrSVCD, 2024-10-02:
[image]
My first build. It's needs 5v for the analog switch chip and I have had bad luck with power banks giving noisy output so I went with 6 ni-mh and a LDO linear regulator.
The next one is going to be a experiment in Manhattan PCB design.

dawsonjon, 2024-10-02:
Very neat!

bahari, 2024-10-07:
[image]
This is my projects setup. I'm adding RX amplifier and programmable step attenuator hardware. I'm also editing the existing code to include a menu for programmable step attenuator.

dawsonjon, 2024-10-07:
Very cool, really like the step attenuator. How are you controlling it?

JimReagans, 2024-10-07:
This is my2nd build of Jon's picoRX. I'm using a pico2 with an external Tayloe decoder which allows me to mess with filter parameters.
[image]
Cheers, Jim W0CHL

dawsonjon, 2024-10-07:
Nice work!

bahari, 2024-10-07:
> Very cool, really like the step attenuator. How are you controlling it?

Basically the relay controlling 3 attenuator base value which is 3, 5, and 10dB. From the combinations of this 3 value I can get an extra 10, 13 and 18dB. I'm adding 3 GPIO from Pico to control the relay via ULN2803. I modified your code with extra attenuator menu to select which attenuator value to used. This really help when I'm attaching my 20dB RX amplifier for strong signal attenuation process.

penfold42, 2024-10-07:
> Basically the relay controlling 3 attenuator base value which is 3, 5, and 10dB. From the combinations of this 3 value I can get an extra 10, 13 and 18dB. I'm adding 3 GPIO from Pico to control the [...]

Nicely done! Are you able to share your code?
Adding support for a PE4302 attenuator is on my list and it would be good to support multiple attenuator approaches.
I'll probably start with using 3 GPIOs to control the high order bits but move to I2C via a PCF8574

bahari, 2024-10-08:
> Nicely done! Are you able to share your code? Adding support for a PE4302 attenuator is on my list and it would be good to support multiple attenuator approaches. I'll probably start with using 3 GPIOs to control the high order bits but move to I2C via a PCF8574
Attached are the code. Effected code are ui.h, ui.cpp, rx.h and rx.cpp. Yeah using i2c is the right choice to control it.
[code-atten-cntrl.zip](https://github.com/user-attachments/files/17292027/code-atten-cntrl.zip)

penfold42, 2024-10-08:
> Attached are the code. Effected code are ui.h, ui.cpp, rx.h and rx.cpp. Yeah using i2c is the right choice to control it. [code-atten-cntrl.zip](https://github.com/user-attachments/files/17292027/code-atten-cntrl.zip)

Thanks! I got a shock when i diffed it against my working tree.
Then realised your code is based on master, not the testing branch where heaps of work has been happening lately! :)

bahari, 2024-10-08:
> Thanks! I got a shock when i diffed it against my working tree. Then realised your code is based on master, not the testing branch where heaps of work has been happening lately! :)

Is there is any major update that improve the receiver performance on the testing branch?

MrSVCD, 2024-10-08:
> Basically the relay controlling 3 attenuator base value which is 3, 5, and 10dB. From the combinations of this 3 value I can get an extra 10, 13 and 18dB. I'm adding 3 GPIO from Pico to control the [...]

How does one build a attenuator?
I don't really have any strong signals nearby but I am going on a vacation to a place should have stronger signals.

penfold42, 2024-10-08:
> How does one build a attenuator? I don't really have any strong signals nearby but I am going on a vacation to a place should have stronger signals.

It can be as simple as 3 carefully chosen resistor values as described [here](https://www.electronics-tutorials.ws/attenuators/pi-pad-attenuator.html). 
In @bahari design this is built on a relay that can switch them in or bypass them.
Then repeat this 2 more times with different values for different levels of attenuation.

You could just pre-build some plug in modules of fixed attenuation and manually plug them in

bahari, 2024-10-09:
> It can be as simple as 3 carefully chosen resistor values as described [here](https://www.electronics-tutorials.ws/attenuators/pi-pad-attenuator.html).
> In @bahari design this is built on a relay [...]

It is based on PI attenuator topology. To calculate it  based on your need, please go to this calculator link: https://leleivre.com/rf_pipad.html

penfold42, 2024-10-19:
Finally got around to ordering better parts and built a new analog board.
74CBTLV3253 and MCP6022 as originally spec'd.

[image]
[image]
I've got enough room for an LNA but might need a 3rd board for band filters.

Performance is somewhat better based on a simple reported signal level using a TinySA as a signal generator.
[image]

Let me know if anyone wants me to do a comparison test of anything else before I recycle the old board...

MrSVCD, 2024-10-19:
Have you tried to battery powered your board? I have noticed that USB power is quite noisy. Might just affect my build since I use a 5v powered switch.

penfold42, 2024-10-19:
> Have you tried to battery powering your board? I have noticed that USB power is quite noisy. Might just affect my build since I use a 5v powered switch.

Yep - no perceptible difference from my laptop.
What is crap tho is a marketing event power bank - very noise switching in it.

mryndzionek, 2024-10-19:
> What is crap tho is a marketing event power bank - very noise switching in it.

Curious, I actually built-in a power-bank module into mine and I almost don't see a difference. I believe it's 134N3P-based.

MrSVCD, 2024-10-19:
> > What is crap tho is a marketing event power bank - very noise switching in it.
> 
> Curious, I actually built-in a power-bank module into mine and I almost don't see a difference. I believe it's 134N3P-based.

Where do you take your power from? directly from the battery or from the 5V?

This module, Lipo Amigo Pro, should not produce any noise [https://shop.pimoroni.com/products/lipo-amigo?variant=39779302539347](https://shop.pimoroni.com/products/lipo-amigo?variant=39779302539347) since it is taking the power from the battery almost directly, only going thru an battery protection circuit.

MrSVCD, 2024-10-19:
I could not help my self, so I built a second one. This is my first Manhattan build.

[image]
[image]

Since I like woodworking I am also building a bookshelf shell for it with built in speaker. The shell is not ready for showing of just yet.

MrSVCD, 2024-10-21:
As part of my build I made a minimal Pico Holder [https://github.com/MrSVCD/PicoRX-accessories/tree/main/PicoHolder](https://github.com/MrSVCD/PicoRX-accessories/tree/main/PicoHolder)

[image]

indrasung, 2024-11-15:
Finally done.
[image]
[image]

dawsonjon, 2024-11-15:
Looks great 👍

MrSVCD, 2024-11-15:
Very clean, very demure as the kids say.

ericeness, 2024-11-16:
Ever since I saw a writeup of this project in Hack-a-day, I have wanted to try it out.  So far I have built two versions and both work to a certain degree.  Each build has been a learning experience.  I am already thinking of V0.3.

[image]

V.01 was built on a breadboard.  I was so eager to try it out that I got sloppy towards the end.  Once I figured out I forgot the connect the 3.3v from the Pico to the plus rail.  It worked!  All I could hear were some AM stations but, the crystal radio has been reinvented.

[image]

V0.2 was soldered down on a protoboard.  I decided to copy Jon's layout as best I could from the various pictures I could find.  I tried to build a nice little control panel but the display didn't work.  I put a new display on a solderless breadboard and yet another Frankenstein monster was born.  This version can receive some shortwave broadcasts but, some of the stations, even with a good signal strength, cannot be received.

[image]

I happen to have an AirSpy HF SDR running across the room.  I use the AirSpy to spot stations to try out on the PicoRx.  I saw that the WWV time station on 15 MHz was coming in good and so I tried it on the PicoRx.  I could not hear WWV on the PicoRx at all. Looking over at the AirSpy, I see three strong signals from the Pi Pico.  Clearly these signals are the local oscillator and perhaps sub-harmonics?  I now regret tucking the wires carrying the LO signals under the Pi Pico to reduce the number of flying wires.  For V0.3 I think I will use twisted pairs to carry the LO and Sampled signals.

indrasung, 2024-11-16:
Hello Ericeness, good works. 
Looking forward hearing from you about using twisted cable from pi pico pin 1 and 2. How about using shielded coaxial rg174.

JimReagans, 2024-11-17:
My pico-rx build is taking shape!  I need to cut down the length of the case yet.
Your mileage may vary,
Jim W0CHL
[image]

MrSVCD, 2024-11-20:
@JimReagans how did you get the colour spectrum waterfall?

vu2iia, 2024-11-26:
Hi all,
I wired pico_rx with whatever components I could find & it's working fine. I have attached photo/screen dump, rx video of 20m & 40m. The setup is powered from a USB port, the antenna is a dipole & no rx filters are used.

For some reason, I see a hump in the scope not a flat noise floor. How to add spectrum scope to the main screen?, instead of having a big s-meter bar-graph, ie as seen in the older version. It would be nice to have a different type of main screen which a user could select. When I change frequency, I can hear chaffing like si4732/35. Is it due to pico cpu load?

Thanks to @dawsonjon  & all other contributors for the interesting project.
mahesh-vu2iia (MK69)
[image]
[image]
[image]

https://github.com/user-attachments/assets/320ab415-96ab-4b02-a078-2fcb25290a61

https://github.com/user-attachments/assets/985a3a1c-c8b4-493e-b893-08da24ee8bb6

vu2iia, 2024-11-26:
> > I see a hump in the scope not a flat noise floor.
> 
> Try the `testing` branch and then you won't complain 😉
yes this main screen looks good.
Thank you,
[image]

indrasung, 2024-11-28:
Hello mryndzionek,  where is the "testing" branch exactly ? Does it has a uf2 firmware for the last display mentioned above?  
Thanks

mryndzionek, 2024-11-28:
> Hello mryndzionek, where is the "testing" branch exactly ? Does it has a uf2 firmware for the last display mentioned above? Thanks

The testing branch - https://github.com/dawsonjon/PicoRX/tree/testing . And no, it doesn't support the TFT display.

indrasung, 2024-11-28:
Hello mryndzionek
Where is the uf2 firmware for this oled screen ? I could not find it. Thanks.
[image]

mryndzionek, 2024-11-28:
> Hello mryndzionek Where is the uf2 firmware for this oled screen ? I could not find it. Thanks. ![Screenshot_20241128-221450](https://private-user-images.githubusercontent.com/63222819/390835779-cbd58afc-593c-4cff-8e72-f250eaf5969a.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI4MDc4NzQsIm5iZiI6MTczMjgwNzU3NCwicGF0aCI6Ii82MzIyMjgxOS8zOTA4MzU3NzktY2JkNThhZmMtNTkzYy00Y2ZmLThlNzItZjI1MGVhZjU5NjlhLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTI4VDE1MjYxNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRlZjZjZjMxZmEwMWI2YzM2NmMxODRmYjQxYTUyNmU4ZWQ0NGZlYmFiZjRmZTAwOTVkNGUyMTg4MTAxZWQwY2MmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.zYQ3JIEydfazVBd2E7Qc25zF1sNnPDqcdljEUTWDYYo)

okay, you mean the **screen**. It should on `testing`.

vu2iia, 2024-11-29:
@indrasung 

This is what I compiled from testing branch, hope this picorx_testing.uf2 helps, it doesnt have many features.
[picorx_testing.uf2.zip](https://github.com/user-attachments/files/17953816/picorx_testing.uf2.zip)

indrasung, 2024-11-29:
VU2IIA thank you.

letonton, 2024-12-03:
[image]

letonton, 2024-12-03:
[image]

dawsonjon, 2024-12-04:
Impressive! Would love to know more about the technical details ...

letonton, 2024-12-04:
Thanks, Jon
nothing special AD607 allows a very low BOM and simple construction
I put a 10,7 mHz xtal filter, the IQ outputs feed directly pico inputs and
one pico IQ out drives AD607 IQ stage local oscillator input.
Only "drawback", 607 is smd only, but as a 0.65 bsc case is not so terrible
to mount.
Currently LO is provided by a lab generator, picoRx code should be adapted
to a ADF synth, or the like, to reach 70 cm band.

letonton, 2024-12-06:
[image]
60 meter band essential and compact transceiver.
I noticed someone suggesting MCP 2036 as front end, tried and works good.
Now I am working on a class-e PA
Next step the transceiver into one Pico.

dawsonjon, 2024-12-07:
Nice, keep us posted!

Tom-ak2b, 2024-12-13:
This is my first version of the Pico SDR using a pico 2. I had some issues with sensitivity so I added an amplifier to the input to get it to work. I'm not sure what the problem is but before I could resolve it my power supply regulator failed (on the end of the breadboard) and took out the pico. I'm waiting for another Pico 2 (I know, who just buys one!?).
The photo is of another Tayloe receiver by Jim Harvey given to me from N1GMM that I will try when the new Pico shows up.
But, before it failed I did a video.
https://youtu.be/AFu9T5pi_aA?si=BRVacuaWyQrXLbQq

[image]

dawsonjon, 2024-12-13:
Nice! I also used the QRP labs Tayloe detector in the first prototype!

CRSourcing, 2024-12-18:
First light :) 
[image]

dawsonjon, 2024-12-18:
Nice 👍

kp4md, 2024-12-22:
This PC Board designed by John Sutter, K6JDS, for the CalQRP Club includes a high pass BCI filter on the input, an audio amplifier and a lithium battery charger module.
[image]

MrSVCD, 2024-12-22:
[image]

My first PicoRX has gotten a new home. I got a laptop charger connected for power and a LM317 & a 7805 for lowering the voltage down to reasonable levels for my antenna amp (12v) and the pico (5v). I had the parts laying around for this approach.
I am going to release the control box after some tuning, adding a bottom piece if nothing else.

@kp4md nice build.

dawsonjon, 2024-12-23:
@kp4md Love the build. I have been following the calQRP builds, looks really good, love all the extra features!

kp4md, 2024-12-24:
@dawsonjon  Thanks for your work and inspiration!

MrSVCD, 2024-12-28:
Upploaded the Contol Box as promised:

[image]
[https://github.com/MrSVCD/PicoRX-accessories/tree/main/PicoRX%20Control%20Box](https://github.com/MrSVCD/PicoRX-accessories/tree/main/PicoRX%20Control%20Box)

dawsonjon, 2024-12-28:
Very smart!

ROBFDP, 2025-01-19:
Hello Jon, here is a picture of my prototype, I don't think it will the construction contest but it does work and should allow me to experiment with the hardware and firmware.  Thank you for making this design available. 

Robert

[image]

dawsonjon, 2025-01-19:
@ROBFDP 
Looks good to me! Thanks very much for sharing :-)

VK2ARH, 2025-02-02:
Although a prototype housing at this point in time with the next version incorporating a number of changes - just wanted to share the housing I designed to hold the CalQRP Club's project board of the PicoRx. I increased the size of the OLED and used the 1.3" variant. This seems to be a neat way of bringing all the components together. I'm very impressed with this little radio, great work Jon and other contributors .

[image]

[image]

[64 more replies]