dawsonjon/PicoRX #107: UI adjustments proposal

mryndzionek, 2024-10-01:
I would like to propose following changes:
 - make the `bigspectrum` screen the default one
 - make the current default screen (`renderpage_original`) a purely status screen, so:
   - remove the scope/panadapter from it
   - divide it into to columns and put all the statuses there potentially also adding nice bars and more info
 - remove the less useful/duplicated screens (`bigtext` and `fun`?)

dawsonjon, 2024-10-01:
I'm definitely open to adding a status screen, the temperature and CPU are probably only of secondary importance. I thought it might be cool to capture the frequency offset from the AMS PLL which might be a useful indication of frequency accuracy, or as a tuning aid.

Rather than make the big-spectrum screen the default, I think we should try to come up with something new, taking the best elements from the existing screens.

I'm slightly torn between:
1. Maybe a 48 pixel high spectrum allowing a little extra space for some compact graphical elements, eg. S-meter/battery indicator.
2. Leaving the spectrum scope off the default screen entirely, and just display the basic info, but make it as simple and visually appealing as possible with large fonts and fancy graphics. Kind of like the big text screen but nicer.

mryndzionek, 2024-10-01:
> frequency offset from the AMS PLL

Great idea!
> Leaving the spectrum scope off the default screen entirely, and just display the basic info, but make it as simple and visually appealing as possible with large fonts and fancy graphics. Kind of like the big text screen but nicer.

I would be for this option.

penfold42, 2024-10-02:
Everyone's a hater.... no "fun" and you're dissin' my C64 inspired status screen :)

Jokes aside, +1 on "make it as simple and visually appealing as possible with large fonts and fancy graphics".

Is is worth adding a config option to en/disable the views that are active?
I suspect if we're only left with 4ish screens its probably not worth it

mryndzionek, 2024-10-02:
> Everyone's a hater.... no "fun" and you're dissin' my C64 inspired status screen :)

I'm not dissin anyone. I love C64. I spent the whole 90's glued to it :smile:

penfold42, 2024-10-02:
> I'm not dissin anyone. I love C64. I spent the whole 90's glued to it 😄

Can you disassemble my avatar ?

mryndzionek, 2024-10-02:
```
* = $0000
INC $D020
DEC $D021
CLV
BVC L0000
```

mryndzionek, 2024-10-04:
With #119 merged I would like to propose following change:
 - `top_menu` is entered on `menu` button long press instead of short press
 - menu short press enables "quick menu" - a bar at the bottom, it the place of spectrum, where `mode`, `bandwidth` and `agc speed` can be quickly  adjusted. Bar disappears on `back` button press, or on timeout.

mryndzionek, 2024-10-08:
In #128 mode control was added after pressing `menu` two times.