dawsonjon/PicoRX #125: Problem with USB audio streaming from core0

mryndzionek, 2024-10-06:
After moving the USB streaming to core0 the problem with mismatch of samplerates seems to be much worse then when streaming directly from core1. First of all I had to decrease the timer task rate from 500us to 250us and increase the ring buffer size, but even then the ring buffer occupancy grows steadily and after a while we're overwriting samples. I tried to measure the mismatch, but it seems it varies with the tuned receive frequency. An adaptive control of rate will probably be needed.

dawsonjon, 2024-10-07:
It is possible to get exactly 15 or 16 kHz. Changing the ADC clock divider to 100 gives an ADC sample rate of 480kHz. This gives exactly 15 kHz with the current decimation of 32 and a slightly reduced CPU loading. Reducing decimation to 30 gives exactly 16KHz but the ADC block size then needs to be reduced to 1920 and the CPU loading is slightly higher.

I think that there will still be some mismatch despite this since clock on the data sink will always be slightly different than the data source, but presume that the USB stack is designed to handle this.

mryndzionek, 2024-10-07:
I managed to implement a PI controller, but the results were still not that good - a lot a wiggle in the weatherfax. I tried switching to 15k samplerate and this seems to be giving good results. PR will land tomorrow.

mryndzionek, 2024-10-08:
Switched to 15k audio in #128.

closed 2024-10-08 by mryndzionek