dawsonjon/PicoRX #126: Fresnel lens OLED magnifier

MrSVCD, 2024-10-06:
I decide to sculpt a 3D print for using a credit card sized Fresnel lens to magnify a 0.96" OLED.
[https://github.com/MrSVCD/CardDisplay](https://github.com/MrSVCD/CardDisplay)
I am working on the documentation but if you can't wait: measure the length between the lens and OLED that is comfortable for you to view and print the back piece closest to you measurement.

I printed first a 80mm version but I got a headache as soon as I looked at it so I printed a 60mm version and the headache disappeared. Another thing I noticed is that reflections in the funnel of the back piece is very distracting so print it in a matte color, I had natural PLA and it works well.

dawsonjon, 2024-10-07:
This is very cool, think I will give it a try!

MrSVCD, 2024-10-12:
I have made a repository to collect links to accessories for PicoRX builds, for example my 3D model mentioned above.

[https://github.com/MrSVCD/PicoRX-accessories](https://github.com/MrSVCD/PicoRX-accessories)

My hope is that people will tell me about their accessories so I can add them to the list.

closed 2024-10-17 by MrSVCD