dawsonjon/PicoRX #127: External down converter support ?

penfold42, 2024-10-06:
Is there any interest in supporting an external down converter ?

I suspect there's 3 levels of "support" we could do

1) configurable frequency offset in the UI
IF inversion to support low side and high side mixing

2) flip LSB and USB if required

3) Control of an external 1st IF
Si5351 over I2C
A separate NCO output

dawsonjon, 2024-10-07:
This is a cool idea. Using an SI5351 would certainly simplify things. You could then tune the Tayloe detector to a fixed frequency that is an exact division of the clock frequency. I'm almost tempted to say that this should be a project in its own right, would be very cool to build a general coverage receiver that goes up to VHF, and it would still be relatively simple.

MrSVCD, 2024-10-07:
One of the things that got me excited about this project is the low component count and the components that you need are quite easy to source aka "a crystal radio for today"
So I say add it as a option but keep supporting the original desig and have fun.

letonton, 2024-12-01:
On GitHub there are Pico libs for si5351 (ex. one by user "kholia").
Additionally we can use Analog ADF435x to reach 70 cm and beyond, SPi
Interface (ex. lib by '"davecrump" also on GitHub)
I agree about a new project thread.

dawsonjon, 2024-12-01:
Yes, I know at least one person has reported success recieving 2m and 70cm using external oscillator and mixer.  I would definitely like to try this sometime.

Maurmari, 2024-12-16:
Mister Kholia's Sî5351 routines work fine, but when I tried to integrate them
into PicoRx I got "undefined reference" errors... the issue is from mix c and c++
codes ? (Frankly I'm a big noob in c !)

dawsonjon, 2024-12-16:
Is it just a matter of using extern "C" {#include "xyz"}?

Maurmari, 2024-12-16:
I am a bit confused about that...

letonton, 2025-01-01:
Someone may help about the matter (compile Sî5351 support into picorx code ) ?

dawsonjon, 2025-01-01:
Sure, happy to help, can you give me some more details?

letonton, 2025-01-01:
Well, like "maurmari" I tried "kholia" code (I work in Win under Visual Studio).
Original code compiles nice, but when I try to add the files to picorx I receive
too a crowd of error messages like undefined reference to the functions.

letonton, 2025-01-01:
Nedless to say, I'm quite a newbie about c/c++ !

dawsonjon, 2025-01-01:
This one? 
https://github.com/kholia/Si5351-Pi-Pico/blob/master/main.c

I don't use visual studio, but it should just be a matter of adding the .c file to the CMakelist.txt. 

```
  add_executable(picorx
      ...
      si5351.c
  )
```

It has already been mentioned that the si5351 library is written in C, whereas PiPicoRx is written in C++.  If you call C code from C++ code you need to wrap the function declarations in an extern "C" {} declaration. This tells the compiler/linker that it is calling a C function rather than a C++ function. You can either add this to si5351.h (I did this in with the SSD1306 library, so you could use this as a template.  https://github.com/dawsonjon/PicoRX/blob/master/ssd1306.h#L36 ). Alternatively you can add the declaration when you include the file like this: 

```
extern "C" {
#include <si5351.h>
}
```
Hope this helps. Cheers Jon

letonton, 2025-01-01:
Bingo ! Thank you very much for the tips.

badwolf3001, 2025-02-08:
Hello, everyone!

I haven't set up my circuit yet, but I'm considering having an option to include a tv tuner as a downconverter (probably with a bypass circuit). The IF output of a tv tuner is usually between 36 and 39 MHz (take 37 MHz as an average). The passband is usually 4 MHz. I believe it will be necessary to mix this output so that it is within the receiver's range (perhaps between 15 and 20 MHz). The frequency control of some tv tuners is controlled via I2C, which would facilitate possible integration. Do you think this approach is feasible?