dawsonjon/PicoRX #135: Build error - testing - pico2-arm

MrSVCD, 2024-10-10:
Hi
I got a build error when I try to build testing for pico2-arm:
```
$ cmake -DPICO_SDK_PATH=~/pico/pico-sdk -DPICO_BOARD=pico2 -DPICO_PLATFORM=rp2350-arm-s ..
PICO_SDK_PATH is /home/mrsvcd/pico/pico-sdk
Target board (PICO_BOARD) is 'pico2'.
Using board configuration from /home/mrsvcd/pico/pico-sdk/src/boards/include/boards/pico2.h
Pico Platform (PICO_PLATFORM) is 'rp2350-arm-s'.
Build type is Release
TinyUSB available at /home/mrsvcd/pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /home/mrsvcd/pico/pico-sdk/lib/btstack
cyw43-driver available at /home/mrsvcd/pico/pico-sdk/lib/cyw43-driver
lwIP available at /home/mrsvcd/pico/pico-sdk/lib/lwip
mbedtls available at /home/mrsvcd/pico/pico-sdk/lib/mbedtls
Using PICO_EXAMPLES_PATH from environment ('/home/mrsvcd/pico/pico-examples')
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mrsvcd/PicoRX/build
mrsvcd@raspberrypi-dev:~/PicoRX/build $ make
[  0%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/mui.c.obj
[  0%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/mui_u8g2.c.obj
[  0%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8g2_arc.c.obj
[  1%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8g2_bitmap.c.obj
[  1%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8g2_box.c.obj
[  1%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8g2_buffer.c.obj
[  2%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8g2_button.c.obj
[  2%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8g2_circle.c.obj
[  2%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8g2_cleardisplay.c.obj
[  3%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8g2_d_memory.c.obj
[  3%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8g2_d_setup.c.obj
[  3%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8g2_font.c.obj
[  4%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8g2_fonts.c.obj
[  4%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8g2_hvline.c.obj
[  4%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8g2_input_value.c.obj
[  5%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8g2_intersection.c.obj
[  5%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8g2_kerning.c.obj
[  5%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8g2_line.c.obj
[  5%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8g2_ll_hvline.c.obj
[  6%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8g2_message.c.obj
[  6%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8g2_polygon.c.obj
[  6%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8g2_selection_list.c.obj
[  7%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8g2_setup.c.obj
[  7%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8log.c.obj
[  7%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8log_u8g2.c.obj
[  8%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8log_u8x8.c.obj
[  8%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_8x8.c.obj
[  8%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_byte.c.obj
[  9%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_cad.c.obj
[  9%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_capture.c.obj
[  9%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_d_a2printer.c.obj
[ 10%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_d_ch1120.c.obj
[ 10%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_d_gp1247ai.c.obj
[ 10%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_d_gp1287ai.c.obj
[ 10%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_d_gp1294ai.c.obj
[ 11%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_d_gu800.c.obj
[ 11%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_d_hd44102.c.obj
[ 11%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_d_il3820_296x128.c.obj
[ 12%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_d_ist3020.c.obj
[ 12%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_d_ist3088.c.obj
[ 12%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_d_ist7920.c.obj
[ 13%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_d_ks0108.c.obj
[ 13%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_d_lc7981.c.obj
[ 13%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_d_ld7032_60x32.c.obj
[ 14%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_d_ls013b7dh03.c.obj
[ 14%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_d_max7219.c.obj
[ 14%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_d_pcd8544_84x48.c.obj
[ 15%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_d_pcf8812.c.obj
[ 15%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_d_pcf8814_hx1230.c.obj
[ 15%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_d_s1d15300.c.obj
[ 15%] Building C object CMakeFiles/u8g2.dir/external/u8g2/csrc/u8x8_d_s1d15721.c.obj
[... 98 lines cut ...]
-- Generating done
-- Build files have been written to: /home/mrsvcd/PicoRX/build/pioasm
[ 44%] Performing build step for 'pioasmBuild'
[100%] Built target pioasm
[ 45%] Performing install step for 'pioasmBuild'
[100%] Built target pioasm
Install the project...
-- Install configuration: "Release"
[ 45%] Completed 'pioasmBuild'
[ 45%] Built target pioasmBuild
[ 46%] Generating nco.pio.h
[ 46%] Built target pico2rx_nco_pio_h
[ 46%] Generating quadrature_encoder.pio.h
[ 46%] Built target pico2rx_quadrature_encoder_pio_h
[ 46%] Building C object CMakeFiles/pico2rx.dir/ssd1306.c.obj
[ 47%] Building CXX object CMakeFiles/pico2rx.dir/picorx.cpp.obj
/home/mrsvcd/PicoRX/picorx.cpp: In function 'int main()':
/home/mrsvcd/PicoRX/picorx.cpp:153:20: error: 'TIMER_IRQ_1' was not declared in this scope; did you mean 'TIMER1_IRQ_1'?
  153 |   irq_set_priority(TIMER_IRQ_1, PICO_HIGHEST_IRQ_PRIORITY);
      |                    ^~~~~~~~~~~
      |                    TIMER1_IRQ_1
make[2]: *** [CMakeFiles/pico2rx.dir/build.make:90: CMakeFiles/pico2rx.dir/picorx.cpp.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:1869: CMakeFiles/pico2rx.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

mryndzionek, 2024-10-10:
> Hi I got a build error when I try to build testing for pico2-arm:
>
> ```
> $ cmake -DPICO_SDK_PATH=~/pico/pico-sdk -DPICO_BOARD=pico2 -DPICO_PLATFORM=rp2350-arm-s ..
> PICO_SDK_PATH is [...]

This is fixed in current `testing` branch.

MrSVCD, 2024-10-10:
Thank you. Now it works on my end too.

closed 2024-10-10 by MrSVCD