dawsonjon/PicoRX #138: List of Alternative Parts?

UtopianElectronics, 2024-10-14:
Hi everyone,

I'm really interested in building this project, but I don’t have much knowledge about electronics/RF, so choosing alternative parts has been confusing, especially since some ICs aren’t available in my local market. Could you kindly provide a list of compatible alternative parts? Also, if using these alternatives requires changes to components like capacitors or resistors, could you let me know what modifications are needed?

Additionally, do I need to have a YouLoop antenna, a preamp, and a wideband LNA? I’m mainly looking to listen to shortwave radio stations right now, so is there a makeshift antenna or a very cheap option that could work for this?

I’d appreciate it if you could also confirm if these recommendations from ChatGPT are correct:

**For the Op-Amp:**
- High Noise Sensitivity: MCP6022, AD8602, NE5532, OPA2330, OPA2228.
- Low-Cost Designs: LM358, LMV358, TL072, TLV2372.
- Low Power/Battery-Powered: TLC272, TLV2372, LMV358, OPA171, OPA2244.

**For the Analog Multiplexer:**
- High-Speed/High-Performance: MAX4619, MAX4785, TS5A23159, MAX4617, TS3A5017.
- Low-Cost General-Purpose: CD4052, DG409, 74HC4052, SN74LV4052A.
- Low ON Resistance: TS5A23159, MAX4785, TS3A5017 (all under 1Ω).
- Low Power Consumption: SN74LV4052A, TS5A23159.

Someone on Reddit also mentioned using an SN74CB3Q3253 for the multiplexer. Could you please confirm this too?

Lastly, if possible, could you give me a list of key specs for any part so that I can show them to a shop owner to help me find alternatives that match the requirements?

Thanks so much!

penfold42, 2024-10-14:
The TL072 and MCP6002 (not the recommended 6022) do work, but their GBP is much lower than the recommended. MCP6022 so U guess you'll lose some of the higher ends of the downconverted signal.

The CD4052 is only good to about 1MHz and the 74HC4052 is ok to about 10MHz.
Their switching speeds are slower which is the limiting factor. 
They also have higher Ron values so you should re-calculate the values of C10,11,12,13 using the formula in the schematic.
In my case with the HC4052 and an Ron of 70 ohms I calculated 27nF

mryndzionek, 2024-10-14:
Regarding opamps, if price is the main concern, choose a 5V one. NE5532P should work and is cheap and popular. I tried LM358P and it works with just a some audible degradation.

> do I need to have a YouLoop antenna, a preamp, and a wideband LNA?

From my experience I would say definitely yes. YouLoop gives weak signal, so LNA is necessary.

dawsonjon, 2024-10-14:
There is a bit of info on op-amp selection here: https://101-things.readthedocs.io/en/latest/breadboard_radio.html#operational-amplifier-alternatives. Basically you need less than 8nV/sqrt(Hz) noise voltage density and better than 10MHz gain bandwidth product. If you are using an LNA you could get away with a lower spec.

For the analog mux the on resistance and switching time are the key specs. The 74cbtlv3253 has an on resistance of 5 ohms and a switching time of 5.5ns. Something in this region would be ideal.

I can recommend the You-loop and LNA, have had good results using this setup. If you want a cheap antenna option, a DIY You-loop can be made from a loop of coax cable and a ferrite core transformer.

mryndzionek, 2024-10-14:
> For the analog mux the on resistance and switching time are the key specs. The 74cbtlv3253 has an on resistance of 5 ohms and a switching time of 5.5ns. Something in this region would be ideal.

So SN74CB3Q3253 is also good - Ron=4ohm and switching times 5.9ns (max).

penfold42, 2024-10-15:
Be careful if you use 5v Op-amps !

The pico is a 3v3 device so you'll need a resistive divider to drop its 5v output to 3v3

mryndzionek, 2024-10-15:
> The pico is a 3v3 device so you'll need a resistive divider to drop its 5v output to 3v3

Maybe it would be good to show the exact schematic. I guess it's a biasing circuit, right? A 100nf capacitor in series + a 50/50 divider connected to 3.3V.

UtopianElectronics, 2024-10-15:
Thank you. I already have LM324N and LM358N chips lying around from years ago. I’m guessing the LM324N doesn’t meet the required specs, but would the LM358N work well enough? I can’t find any LNAs here, except for a module marketed as "NRF24L01+PA+LNA," which seems to have an LNA chip onboard. Is there any way I could manage without an LNA, or possibly even without using a YouLoop antenna altogether?

mryndzionek, 2024-10-15:
>  but would the LM358N work well enough? 

For listening to "commercial" AM stations with fairly strong signals should be okay. You can always use DIL/DIP connector and test different opamps later, as you can get them.

> I could manage without an LNA, or possibly even without using a YouLoop antenna altogether?

If you live in a /RF quiet/rural/ environment you can use a ordinary wire antenna. The signal will be stronger. I live in a noisy environment and don't have space for a wire antenna and magnetic antennas (like youloop) ale the only option. MMIC LNA's, often old and obsolete, can be found in shops for little money. You can also look into building an active loop with a simple transistor amplifier, like https://www.pa1m.nl/simple-active-receive-loop/.

UtopianElectronics, 2024-10-16:
> You can also look into building an active loop with a simple transistor amplifier, like https://www.pa1m.nl/simple-active-receive-loop/.

Still doing some research... I live in an isolated country at risk of war, so I’m trying to build this to receive SW radio for listening to news stations abroad, in case our electricity infrastructure gets hit. There’s no access to international financial/postal services here, so I can’t order an LNA from abroad, and even if I could, they’d be extremely expensive. 

I live in a city and need a small, portable antenna I can move around with. The active loop with a simple transistor amplifier you linked needs 12V to operate, right? For now, I just need something basic for receiving MW/SW stations between 1.5 MHz and 18 MHz (In case you’re wondering why I’m not just building a SW receiver, I couldn’t find variable capacitors, toroids, etc. either.) Is there any way to compensate for the lack of an LNA and a You Loop antenna? What about using a coil-loaded antenna?

mryndzionek, 2024-10-16:
> What about using a coil-loaded antenna?

I guess it's hard to say without trying. In my experience YouLoop behaves very well in presence of noise. If there is not much noise than other antennas might perform better. If you're interested in specific stations/frequencies than a tuned magnetic loop might be good. See [here](https://www.youtube.com/watch?v=jlLj6hvG-2c). Here is a link to calculator - https://miguelvaca.github.io/vk3cpu/magloop.html . You can roughly see how many turns and what diameter covers which frequencies.

UtopianElectronics, 2024-10-18:
I’m not sure if I can complete this project due to the lack of RF parts here, but I do have some common equipment for receiving satellite TV, including a few LNBs and a receiver box. Is there anything inside them I could salvage that might be useful for this project?

penfold42, 2024-10-19:
You might be able to grab the RF front end from a 27Mhz CB radio or a 1980s shortwave radio.
The 74HC4052 I was using came from the tuner board of a pace set top box.
Most of the satellite stuff will have parts designed for frequencies much higher than HF

yasin112358, 2024-12-14:
what about miniwhip antenna does anyone used it?

Tonytti, 2025-02-28:
I've done a test with 74hct4053 and it runs beautifuly

mryndzionek, 2025-04-29:
Has anyone tested the following opamps?
TLC082
MC33078
TS522IDT
LM833
NJM4580