dawsonjon/PicoRX #141: Loop antenna build questions

MrSVCD, 2024-10-21:
I am looking to build a active loop antenna [https://www.pa1m.nl/simple-active-receive-loop/](https://www.pa1m.nl/simple-active-receive-loop/) and I have some questions:

1. Can I use steel for the loop instead of aluminum?
2. Can I use a FT-50A-43 core instead of a FT37-43? [https://www.electrokit.com/en/ferritkarna-ft-50a-43-7.2mm](https://www.electrokit.com/en/ferritkarna-ft-50a-43-7.2mm)
3. Can I use 9v instead of 12V?

Schematic for those that don't like following links:

[image]

penfold42, 2024-10-22:
I'm no expert but some thoughts come to mind:

>     1. Can I use steel for the loop instead of aluminum?
Google doesnt seem too positive on steel. lower conductivity, heavier (and rustier!)

>     2. Can I use a FT-50A-43 core instead of a FT37-43? https://www.electrokit.com/en/ferritkarna-ft-50a-43-7.2mm
You might need to adjust the number of turns if the inductance is critical

>     3. Can I use 9v instead of 12V?
You'll lose some gain and might also have to adjust the bias resistors

MrSVCD, 2024-10-22:
I found aluminium locally so that is no longer a problem. A hardware store for building had it as part of the tile accessories.

MrSVCD, 2024-11-02:
I have now made the amplifier and purchased a aluminum strip.
[image]
[image]
[image]
[image]
There is a high possibility that my transformer doesn't work but I wanted to try if it works with cat 6.

penfold42, 2024-11-02:
The links to images dont work either! :)

MrSVCD, 2024-11-02:
@penfold42 The latest jpeg encoding tricks in the gnu imp did not work.

MrSVCD, 2024-11-02:
It works! And it has better reception than my wire antenna.
Pictures of the success will come tomorrow.

MrSVCD, 2024-11-03:
Pictures as prommised:
[image]
[image]
[image]

Here is a video of me turning the antenna back and forth to show of the directionality of the antenna: [RadioTest-640.webm](https://github.com/user-attachments/assets/a6cc77d9-710a-46ea-b00c-a3165e0f1bd1)

Ps. This is inside a metal clad house, so it is going to be better when I get it outside.

closed 2024-11-03 by MrSVCD

dawsonjon, 2024-11-03:
Nice, really neat looking antenna, thanks for sharing!

penfold42, 2024-11-03:
Nice work!