dawsonjon/PicoRX #146: Attenuator design

MrSVCD, 2024-11-08:
I am trying to make a attenuator box but I am a bit stumped on getting the correct values unless I use trim pots.
Is this good enough?  
[image]

I just realized as I am writing this that I only need 5dB, 10dB, 20dB so it is not that expensive to buy trim pots. So what I am asking if I should build it that way.

penfold42, 2024-11-09:
Too funny - was looking at a pi calc website last night - in my case for the downconverter I want to play with.
You can combine resistors in series or parallel to get the target values. I too cant seem to find my 17.615 ohm resistors...
The other option is just pick value that are close enough - I doubt you'd be able to hear or measure any real difference.

Can I interest you in a PE4302 that's on my list to add support for?
It just take a 6 bit binary input to select from 0.5 to 31.5 dB attenuation.
$5 and a single wire would let manually you select 0.5, 1, 2, 4, 8, 16 dB
two wires gets you some combinations - 8+16 = 24dB

MrSVCD, 2024-11-09:
Found the page for the replacement IC [https://www.psemi.com/products/rf-attenuators/general-purpose-rf-digital-step-attenuators/pe4312/](https://www.psemi.com/products/rf-attenuators/general-purpose-rf-digital-step-attenuators/pe4312/)

And also I found EU listing for PE4302 on ebay. 3v supply is going to be a challenge... * Reading data sheet * It is not going to be a problem since CR2032 exist and it draws 0.1mA.

MrSVCD, 2024-11-09:
> As for integration with Pi Pico - My plan is to couple it with a LCD I2C backpack so we can use I2C to set the pins

I was a bit confused what you meant but you mean a add-on board for a parallel LCD so it can be used with I2C like this one: [https://www.electrokit.com/en/i2c-interface-for-lcd](https://www.electrokit.com/en/i2c-interface-for-lcd) (They have also this one: [https://www.electrokit.com/en/i/o-expander-8-kanal-i2c](https://www.electrokit.com/en/i/o-expander-8-kanal-i2c) which uses the same IC)