dawsonjon/PicoRX #148: FT8 Decoding on the TFT display

charudatt, 2024-11-23:
Now that we have the TFT display add-ON can we decode FT-8 on it directly ?
de VU2UPX

dawsonjon, 2024-11-24:
This is something I would like to try. There are a few projects out there that can decode FT8 with a pi-pico, so we know that it's possible. The question is whether there is enough spare resources to receive and decode. The code I have seen is using floating point arritetic.for the LDPC.decoder, so I think it could be made to go faster using fixed point approximations of tanh. I think memory would be critical, decoding FT8 involves storing 12 seconds worth of signal, which is uses a fair chunk of the available RAM. A Pico2 is about twice as fast and has twice as much memory, so should solve both problems.

charudatt, 2024-11-24:
thanks for the promising reply. Looking forward to hear from you on this. Thanks for sharing your experience with us. I have designed a PCB for the project, would be submitting pictures of the same when I assemble and test one.

closed 2024-11-24 by charudatt