dawsonjon/PicoRX #150: Modification

letonton, 2024-11-30:
Hi all !
Not exactly an issue, but...
I'd like to add an additional frequency set method
I have four preferred frequencies, I want to quick switch between them
but keep the encoder.
Suggestions?
Thank you !

dawsonjon, 2024-11-30:
Sounds intriguing, perhaps you could tell us a little more about what you have in mind? Would storing those 4 frequencies in memories work?

letonton, 2024-11-30:
It's a 60 meter transceiver intended for field usage.
The range have four standard channelised frequencies, I guess is useful to
quick switch them with a rotary switch.
The rotary encoder is still necessary to tune different frequencies.
I wonder if a single transceiver is possible with Pico 2...

dawsonjon, 2024-12-01:
There's a shortcut to the memory load function if you press the encoder, if you then put the 4 standard channels in memory you would have fairly quick access without having to change the software. Having said that it would be fairly easy to add a rotary switch or some extra buttons with a fairly straightforward mod to the software. I think a transceiver is possible with pico2 (or even pico1 potentially).

letonton, 2024-12-01:
Thank you.for provide the mod !

dawsonjon, 2025-01-03:
Very cool, will.take a look!