dawsonjon/PicoRX #152: fldigi and PicoRX

MrSVCD, 2024-12-01:
Hi,
I am looking into making a WEAFAX receiver to have running for a bit, because fun.
I have trouble getting fldigi (4.2.06) to comunicate with my PicoRX running the latest stable release (2024-11-21).
Has anyone gotten it to work?

MrSVCD, 2024-12-15:
[image]
I got it sort of working with digital ducktape.
I still can't get fldigi to talk directly with PicoRX.

dawsonjon, 2024-12-15:
Cool, looks good. Can you give me any more details about the issue? From the image, I'm guessing that cat control is working with grig, but not fldigi? I haven't tried cat control in Fldigi,  It might be that it's using a cat command that I haven't implemented. Looks like the USB audio is working OK? On a Linux platform?

MrSVCD, 2024-12-15:
For some reason when I press initialize in fldigi, fldigi freezes for a bit and then unchecks hamlib. I have not found the log yet so I don't have more info.

But receiving weafax works great. 
[image]
(The jpeg is quite compressed to fit the gemini BBS file size limit)

closed 2024-12-28 by MrSVCD