dawsonjon/PicoRX #155: noise and background sounds

yasin112358, 2024-12-14:
ı have finally made my picorx but the noise and background sounds are really high ı did not use band pass filter and ı used miniwhip antenna in stead of dipol. can you suggest anything about it what would be the problem?

[image]
[image]

https://github.com/user-attachments/assets/4cc801f4-f224-4623-b202-15146dc0856f

dawsonjon, 2024-12-15:
Hi,
Build looks good, I particularly like the neat wiring that always turns at right angles.

With regards to noise there are a few possibilities.

The HF bands a generally quite noisy, so its informative to compare the noise level with and without the antenna connected. Ideally, the noise floor as indicated by the spectrum scope/s_meter should increase slightly when connecting the antenna. If it does, that's good news! It means that noise from antenna is larger than any internally generated noise within the receiver, and the antenna noise is large enough for the ADC to see. A Low Noise Amplifier can help with this, I find that combining an LNA with a You-Loop antenna gives pretty good results. I haven't tried the mini-whip, but I assume that it includes an LNA.

Sometimes external noise sources can have a negative impact on the receiver performance, its worth trying to switch off electrical appliances to eliminate them as a source of interference.

Both the TFT and OLED display can generate quite a bit of noise, its worth trying to disable them by setting the "timeout" in the HW menu. I haven't been able to completely remove the display noise, but it can be improved by careful layout. You might see some improvement by adding a ferrite bead to the display connections.

There can be some noise from the PSU too. Using series inductors and parallel capacitors can help with this (I see you have these). Its worth experimenting with the quantitiy, size and location of capacitors too. With careful construction/filtering I find that switched-mode  3.3v output from the pico is clean enough to get good results. You can eliminate the PSU as a source of noise by temporarily supplying the receiver components from a battery or other low-noise power source.

Another source of noise is switching currents from the PWM and Local Oscillator. I found that using a second inductor to provide a separate supply to the op-amp and input bias circuit (but not the analog switch) can help prevent LO switching noise from getting into the op-amp power supply. 

One downside of breadboard construction is the lack of ground plane, it is possible to make significant improvements by giving careful thought to the ground connections. The biggest sources of noise are the local oscillator connection to the analog switch and the PWM connection to the headphones (and their respective ground return paths). The parts of the circuit most susceptible to noise are the inputs to the op-amp. 

I found that providing a dedicated ground connection from the pico analog ground to the ground pins of the op-amp/sampling-capacitors/input bias-network and antenna ground (but not the analog switch) prevents any of the switching return currents flowing in the analog ground path. When I tried this I got quite a significant reduction in the noise floor with no antenna connected.

yasin112358, 2024-12-20:
thanks a lot, ı will check them out.

asflorenciotec, 2024-12-24:
Put a 1000uF 6,3v eletrolic capacitor near OLED display, kill the noise oled noise.

yasin112358, 2025-01-04:
I will try it thank you Jon .

Tonytti, 2025-01-15:
If you don't have input passband filters you cannot use a miniwhip, because is a broadband antenna. I use a magnetic loop without filters, because the hi q of it, reception is excellent.