dawsonjon/PicoRX #157: Pico2

letonton, 2025-01-01:
Hello
how to correctly compile uf2's for pico2 ?
I compile no problem Rp2040 on Win and Visual Studio.
Pico 2 compiles as well, but the resulting uf2's are rejected when loading them...
Thanks and have a fine new year.

dawsonjon, 2025-01-01:
Ok, happy to take a look, there are some generic instructions to build from the command line here: https://github.com/dawsonjon/PicoRX?tab=readme-ov-file#build-projects. Does that give you the same issue?

letonton, 2025-01-01:
Yes I guess is a matter of cmake configuration...
Someone uses Visual Studio ?

Tonytti, 2025-01-15:
I had the same problem, i solved it by downloading bynary codes directly into pico2. Visual studio gave me lot of issues I didn't have with pico1.

letonton, 2025-01-15:
Yes, I installed nice the SDK for pico1 fom the installer pointed out by Jon.
But there is no more an installer to update to Rp2350 :(

BrownBear72, 2025-05-23:
Hello!
Please help. I can't load a file into pico2. I took the sources from different branches. Everything works for me only with PicoRX-testing. I get a uf2 - picorx file (483k). But it doesn't work on pico2. I think I get a file for pico, not for pico2. How can I get a file uf2 for pico2 in  Pico Visual Studio ?

dawsonjon, 2025-05-23:
@BrownBear72 Hi, there isn't really anything useful on the testing branch. The master branch has stable code, and the tft branch is being actively developed.

I can't really advise you on VSCode, I just build using CMake from the command line. There are instructions here: https://github.com/dawsonjon/PicoRX/blob/master/README.md#build-projects

For pico2, you just need to add a couple of extra options when you run CMake. If you have already built for the Pico, you will need to delete the build folder and rerun CMake.

There are some USB issues in the latest version of the pi Pico SDK, so you will probably need to roll back to an earlier version to get it working, but sounds like you are having a build issue rather than a run-time issue.

dawsonjon, 2025-05-30:
Honestly I don't know. I was planning to stick with an older version of SDK until it catches up.

dawsonjon, 2025-05-31:
I'm working on the next release, but it's not ready yet, I have some issues that I need to work through and limited time. Please be patient, it's coming soon.

SebKuzminsky, 2025-11-19:
I looked into the tinyusb issue a bit and discovered the following:
| pico-sdk version | uses tinyusb version | works? |
|-------------------------|------------------------|------------|
| 2.1.0                     | 0.17.0                  | ✔️         |
| 2.1.1                     | 0.18.0                  | ❌         |
| 2.2.0                     | 0.18.0                  | ❌         |

When built with either of the two versions of pico-sdk that use tinyusb 0.18.0, the PicoRx firmware crashes the pico at boot, then the watchdog reboots the pico and we're in a boot loop.

Building with pico-sdk 2.1.0 produces working firmware.

Building with pico-sdk 2.2.0 but with the tinyusb submodule at 0.19.0 (instead of the default 0.18.0) also works.

Hope this is helpful.