dawsonjon/PicoRX #158: BOM?

Tom-ak2b, 2025-01-11:
I am not sure if this is the place to ask questions but if not let me know.
I have a bom  that leaves open exactly what IC's to purchase. Is there a bom that lists part numbers from Mouser or Digikey or any other source that I could use.
Thanks, 
Tom ak2b

dawsonjon, 2025-01-11:
Did you spot this parts list? It includes digikey links for the two ics. I didn't put links for the resistors and capacitors, I figured it was probably easiest to order a selection pack if you don't already have them.

Cheers
Jon

https://101-things.readthedocs.io/en/latest/breadboard_radio.html#parts-list

Tom-ak2b, 2025-01-11:
You mean did I spot in plain sight? Sorry, missed it.
Thanks, it is exactly what I was looking for.

closed 2025-01-11 by Tom-ak2b