dawsonjon/PicoRX #160: TFT Spectrum Display is not right

ericeness, 2025-01-20:
I am working on v0.3 of my PicoRX hardware.  This version is built on a solderless breadboard with an improved layout, but that is not my issue.

I have noticed that the spectrum display at the receive frequency is duplicated at the -10 KHz portion of the display.  At the +10 KHz the spectrum is deleted.  For my picture I used a local AM radio station but I have observed the same thing on 40 Meter SSB.  I can hear 40 Meter SSB with the mode set to LSB, but I may have the I/Q sides swapped.

Could a wiring mistake on my part cause this artifact on the display?

[image]

dawsonjon, 2025-01-21:
@ericeness 

Hi, I think theres more than one thing going on here.

1. Yes there is a known issue where the last approx 6kHz of the display wraps. It's an artefact of the IF frequency shift which is circular for discrete signals. There is some discussion about it here.

https://github.com/dawsonjon/PicoRX/issues/52#issuecomment-2341739419

2. Note that the duplication you are seeing is not caused by the issue above, the duplicate peak is an image. This suggests that either your I or Q channel has become disconnected which means the receiver can no-longer suppress images causing the spectrum to be mirrored.

3. The colours in the display are reversed. There is a menu option under hardware config to reverse this.

Cheers
Jon

ericeness, 2025-01-24:
> 2\. either your I or Q channel has become disconnected

Jon, thank you for your sage advice.  I got out my oscilloscope and sure enough, one of the opamp outputs was flatline.  I had the bright idea of using half of the mux for the I samples and the other side for the Q samples.  I connected both inputs together to make sure the same antenna signal reached both sides of the mux.  This approach shortens the analog path and simplifies the layout on a breadboard, or so I though.  I just don't get why only one half works.  I rewired to use only one half of the mux and boom, it worked as expected.

73, Eric

dawsonjon, 2025-01-24:
@ericeness great, glad you got it sorted!

penfold42, 2025-01-24:
> This approach shortens the analog path and simplifies the layout on a breadboard, or so I thought.  I just don't get why only one half works.
> 
I took this approach with my second build - photos in the show us your build issue.
Is there a chance you had the unused outputs shorted ?

ericeness, 2025-01-25:
@penfold42  
> I took this approach with my second build - photos in the show us your build issue.

Thank you confirming that my approach should work.  I took another pass at debugging.  I started moving the sample capacitors from one half of the mux to the other.  Once I found the capacitor that was not getting signal and confirmed that the wireless breadboard making a good connection, I concluded the issue was a bad solder joint on the TSSOP to dip adapter board.  I resoldered the pins on one half of the mux part and problem solved.  Now I can go back to my noise reduction experiments.

Thanks for the help,  Eric