dawsonjon/PicoRX #163: Can't upload new frequencies

f4goh, 2025-02-24:
Hello,
I try to use upload_memory.py
COM port open but it seems wait ser.readline()
i select *USB Memory Upload* menu item in a submenu is ok

```py
with open(filename, 'rb') as input_file:
      for channel in buffer:
        for location in channel:
          ser.write(bytes("%x\n"%(location), "utf8"))
          ser.readline()   #wait ack ?
      ser.write(bytes("q\n", "utf8"))
```
Perhaps i miss something ?

73

dawsonjon, 2025-02-24:
@f4goh No, I don't think you are missing anything. What platform are you running on?

f4goh, 2025-02-24:
Hello,
Thanks for answer,
Linux Mint, with python3.8 and pyserial by pip install
serial /dev/ttyACM0
I also post my project here. 
https://github.com/f4goh/SDR-PICO
Did you know is there any driver to detect usb sound card from pico for my old Win7 OS (4011) ?
but usb RPI mic sound card works on Linux Mint :)

dawsonjon, 2025-02-24:
Thanks, I will take a look into the memory writing, I checked it was working recently (also on Linux mint) but that was on a newer branch.  It may have broken somewhere along the line. For windows 7 you might need to install a third party uac2 driver, I think there are a few to chose from.