dawsonjon/PicoRX #165: PicoRX hangs when trying to connect to a Linux MX USB port

sweiland-openrails, 2025-03-06:
I have a strange problem. I've yet only build the digital part of PicoRX: the rpi pico, rotary encoder, display and buttons.

I compiled and uploaded pico2rx.uf2. When connected to a Linux MX computer USB port it hangs and the display is showing the PicoRX splash screen halfways. When I just connect it to a phone charger it works ok. I then see the frequency and it changes when turning the rotary encoder. Command lsusb gives a line, so it is connected:

Bus 003 Device 024: ID cafe:4011 dawsonjon PicoRX

When using the pre-compiled uf2 files, it works also ok, both of them. When I compare the downloaded pico2rx.uf2 and the compiled pico2rx.uf2, they both are of the same size. The contents is different.

I see this problem only when connecting to Linux MX. Connecting to Windows 10 is ok.

Anyone an idea what could be the problem?

dawsonjon, 2025-03-06:
I have a very similar issue if I compile with the latest version of SDK. I'm working through it, but don't have a solution yet. For now, the best workaround is probably to roll back to an earlier version of SDK.

sweiland-openrails, 2025-03-06:
Ok, thanks for your quick answer. I'll try a previous version of the SDK.

mryndzionek, 2025-07-29:
New `pico-sdk` was released today, but the issue is still there.

ea2ehc, 2026-08-18:
I was about to give up until I read your solution. Thank you so much for sharing it. With the new TX/RX version and your contributions, it's going to be a great project.

Thanks again, you've made my day.

73 - de Pablo.

mryndzionek, 2026-08-18:
> I was about to give up until I read your solution. Thank you so much for sharing it. With the new TX/RX version and your contributions, it's going to be a great project.
> 
> Thanks again, you've made my day.
> 
> 73 - de Pablo.

Few days ago I made another bump to tinyusb. A PR (#312) updating pico-sdk and setting all the right dependencies versions is pending.