dawsonjon/PicoRX #167: CAT on Windows

bugsel, 2025-03-09:
Thanks for the interesting project, yesterday I assembled the Pico RX, everything works fine, tested it only on FT8, on absolutely unsuitable antenna, on Linux the CAT interface works, but I can't launch it on Windows? The port appears, are there any special drivers needed for it?

[image]

[image]

[image]

bugsel, 2025-03-09:
Place for the OLED display does not match the most popular options that are sold: G-V) I didn't immediately understand, only after I burned my finger on the display)

dawsonjon, 2025-03-09:
@bugsel Hi, on windows you may need to set the force DTR (high) option to get CAT working. You are right about the old connections, I hadn't realised until recently that some olds were the other way round. When I update the PCB I will add some links.

bugsel, 2025-03-09:
@dawsonjon can use any display, there are 2 jumpers to change the polarity, can simply add labels to the pads on the mask

spanglz, 2025-03-16:
@bugsel did you ordered pcb? Are you from RU? And could you give me a contact for PCB ordering?

bugsel, 2025-03-17:
@spanglz I ordered from pcbwave, they accept ru cards on the website, delivery takes about a month, I have 10 of them, I can send one) p.s. same TG, write to me