dawsonjon/PicoRX #175: Using "USB Upload" ( Memory )

klaus344, 2025-04-26:
In belonging "User Manual" under "Hardware Configuration Settings-Part 2" also setting "USB Upload ( Memory ) can be found. This should open an opportunity for uploading new memory files according to personal wishes. 
For first test i only wanted to try uploading unmodified file "memory.py" like it can be found under "utils" in belonging GitHub folder. For this i did select necessary settings until OLED Display did show "Ready for memories", but after this, when trying to use a terminal program ( Baudrate ? ) for uploading file "memory.py ", nothing did happen any more ( to continue restart of PICO was necessary only 3by short power interruption ).
So my question is, what am i doing wrong and what about maybe need of some special program or special parameters for uploading those py-files?
Maybe Jon, you could describe necessary procedure a little bit more detailed.

dawsonjon, 2025-04-26:
@klaus344 Hi, the idea is that the upload_memory.py utility (under the memory_loader directory) can be used to upload a CSV file containing customised memories. First edit the template CSV file, then select the upload memory item in the hardware configuration menu, when the receiver is ready, you can upload the CSV file using the upload_memory.py file. The command is:

python upload_memory.py memory.csv

I have had reports that the script is broken, so it may not work anyway, there is a tested working version in the next release though. Plan is to add a graphical uploader similar to chirp to allow memories to be managed.

I'm also very keen to provide some templates with region specific memories defined, so I would welcome any contributions of CSV files.

klaus344, 2025-04-27:
Great, so we are waiting for what you are planing for the future. Nevertheless i will try uploading memory data like proposed. 
Thank you for your help.

dawsonjon, 2025-04-27:
I checked the released version and it does work using the command line utility, but I hoping to add something a bit nicer when I get some time.
Cheers
Jon