dawsonjon/PicoRX #176: Streaming IQ samples over USB

mryndzionek, 2025-04-26:
I finally have some time, so I started investigating IQ samples streaming via USB. There is a PR with some initial work (#174). TinyUSB is fairly limited when it comes to USB device functionality. It is possible to have only one audio device (UAC2) with one, two or four channels. My plan is to change current USB configuration from mono to stereo and add USB streaming mode setting (accessible via UI/CAT). In "audio streaming" both channels will have the demodulated audio. In "IQ streaming" one channel will have I data and the other Q data. Those this sound okay? Maybe there are other suggestions.

Meantime here is a quick demo:

https://github.com/user-attachments/assets/78792ead-3cdb-40d9-9698-ee10d1b056ec

dawsonjon, 2025-04-26:
Sounds good to me!

mryndzionek, 2025-04-26:
> Sounds good to me!

Okay! Next PR - #177.

mryndzionek, 2025-05-04:
Basic IQ streaming is now on `master`. AGC probably needs some adjustments. There is also no USB volume control yet (a fix to USB volume control needs to be integrated first - #182 ).

closed 2025-05-04 by mryndzionek