dawsonjon/PicoRX #233: PicoRx firmware

backronymm, 2025-09-22:
Hello,
I apologize for the amateur question: 
[Pre-Built Binaries for Pi Pico Rx](https://github.com/dawsonjon/PicoRX/releases/tag/v0.0.1)
the archives contain two files
Why do you need a file battery chek.uf2? 
You can't write more than one file in Pi Pico?? 
Sincerely yours,
 Serg

dawsonjon, 2025-09-22:
Hi Serg,

You don't need the battery check binary.
I had some reports of the battery voltage being reported incorrectly, I
added the battery_check.uf2 as a test build that could be used to check the
battery voltage independently of the full firmware.
Best of luck!

Kind Regards
Jon

backronymm, 2025-09-22:
Thank you very much!

electroheizer-droid, 2026-02-14:
Hello Jon, first of all, thank you for developing the project along with the contributions of other friends whom I also congratulate. Today I set up the project. I know how to program, but I'm finding it difficult to understand how to compile on this platform. I understand Arduino well; perhaps in the future, a basic code can be ported to work from Arduino. I found that this software is missing something: it shouldn't remember the path when entering a menu. If I change, for example, from LSB to USB and then exit, it returns to the initial menu, and I have to search for the specific menu again. If it helps, ATS-Mini incorporates this option. I would have liked to help, but I'm not familiar with the platform. I'm just getting to know the Pi Pico RP2040 board. Using an ST7735 TFT screen might be interesting in the future, and it's lightweight. I've used it with an Arduino Nano, since for now, the waterfall model isn't very useful (I wouldn't have this option in the menu on an OLED screen). Well, the most important thing for now would be to memorize the menu path; it would make accessing the settings much easier. Thanks for listening, and I hope this option can be integrated in the future, perhaps a noise reduction feature—I read something about that somewhere. Greetings from CE2GVT.

electroheizer-droid, 2026-02-20:
> Hi Jon, thanks for replying. Your suggestion to participate would be
> interesting. Two days ago, I was given a DeepSDR 101 SDR clone. Although I
> was able to update it, it leaves much to be desired. It would have been
> interesting to work with this hardware, as many people have bought it, and
> although it works, it's very basic compared to your project. It also uses
> an ARM processor, but I'm not familiar with it at all. However, it has
> potential; I'm just putting it forward as an idea.
>
> Your radio, software, and hardware are excellent. It's been wonderful to
> use it and see the excellent work and time invested. I used an SDR IQ that
> I built for a laptop with a 74HC4053, but it only works up to 14 MHz. I'll
> build the hardware with an SFT3253 later. I've ordered several RP2040
> boards to experiment and learn more about this Raspberry Pi pico. The
> learning curve will be slow, but I hope to succeed. If there are any
> updates on your project, I'd be happy to try it out and contribute ideas.
> Congratulations, Jon from CE2GVT-Gabriel.

hi jon, i send a interesting project
https://github.com/kaefe64/Arduino_uSDX_Pico_FFT_Proj , this project is
make on rp2040 & MBED OS rp2040, dale una ojeada talvez sea
possible rescatar el receptor y/o optimizar el proyecto como un transceptor
sdr directamente, esta interesante pero es un proyecto grande para mis
conocimientos y no muy estetico en pantalla, ademas me dice que se queda
corto de memoria ram.

electroheizer-droid, 2026-02-20:
uSDX_TFT.rar
<https://drive.google.com/file/d/1z3ToMAuJjnOpXRRwrPG5N3yUm3R1RcSi/view?usp=drive_web>
i send this link to download this project